RoboJackets/robocup-software #778: Make a Competition mode?

chachmu, 2016-07-14:
Would it be a good idea to make a mode for competition (maybe "make run-comp") or something that restarts itself if it crashes and maybe defaults to logging somewhere to keep us from forgetting to log? This would require the completion of #762 but it would help solve a lot of our issues with seg faults during competition or other problems.

jgkamat, 2016-07-14:
Once we have arguments that load playbooks and stuff like that, we can just add a target in the makefile with a bit of shell script that would do this for us. It shouldn't be too hard once the actual functionality is implemented.

guyfleeman, 2016-07-14:
Maybe some safety checking like external ref mode and behavior tree view. I know we forgot these sometimes at comp, and I don't think in the heat of the moment people will actually care about a checklist.

chachmu, 2016-07-14:
Behavior tree view and external ref are the default already so when soccer relaunches we wouldn't need to worry about that. The make command would need to take arguments for which side of the field to defend and maybe if we are blue or yellow. I know which team we are is supposed to be encoded in the ref packets but I'm not actually sure if that always gets set properly.

ryanstrat, 2016-07-14:
The ref packets have team names and colors set 100% of the time at comp

jgkamat, 2016-09-12:
Because of segfaults, this is a priority (implemented in some form, be it in c++ or in the makefile). Bumping priority up.

PR #867 (2017-01-27): Make a Competition Mode
This will make it possible to automatically restart soccer if it segfaults during competition as well as set the important parameters automatically. Closes #778.

TODO:

- [x] Set Yellow/Blue Team from refpacket
- [x] Closes #762 
  - [x] Add argument for Defend Vision
  - [x] Add argument for Vision Channel 
- [x] Automate restart of soccer
closed 2017-01-27 by jgkamat