RoboJackets/robocup-software #780: Create a virtualbox builder for robocup-software

jgkamat, 2016-07-16:
With the new recruits coming in in ~2 months, we need to start planning how we're going to onboard them. Since IGVC has had much better success by simply giving out vitualbox images, I'm going to create a virtualbox image builder for robocup-software (to at least provide that option for new recruits).

I'm going to use packer, since I've heard good things about it. I'm probably going to be using this template: https://github.com/boxcutter/ubuntu, forked to RoboJackets.

This is mostly a tracking issue.

chachmu, 2016-07-16:
Would we want to also have ssl-refbox and maybe QTCreator on the image so the new members won't have to worry about any software installation?

jgkamat, 2016-07-16:
good idea! If you have any other peices of software that we should have, let me know. I'm adding a list of software I'll add in OP, let me know if you think of anything else!

Also, one of the downsides of offering this is that we won't be able to have the firmware demo unless enough people install ubuntu on physical hardware...

chachmu, 2016-07-16:
I have never tried but can't you mount usb devices straight to virtualbox?

jgkamat, 2016-07-16:
~~We tried it last semeseter, but windows/osx has to actually recognize the USB device in order for that to work. That means it will likeley work on a linux host, but not on a windows/osx host as they don't allow you to detect custom hardware.~~

Actually, the mbed just shows up as a flash drive, so I think we can do this. I'll include firmware just in case.

jgkamat, 2016-07-23:
This will continue here: https://github.com/RoboJackets/robocup-images

This issue will still track it's creation.

jgkamat, 2016-07-27:
First sample virtualbox file is here: https://transfer.sh/t66bB/robocup-f209e275.zip

I'll keep working on it sometime next week.

chachmu, 2016-07-28:
@jgkamat To confirm that I'm doing this right, I can just stick the .vdk in the SATA for a preexisting ubuntu virtualbox machine and start it up right?

jgkamat, 2016-07-28:
user: vagrant, password: vagrant

You can use file->import appliance to import the app, but swapping in the vdk file should be fine too (but more work)

chachmu, 2016-07-28:
@jgkamat The username and password worked although it just started up a terminal, there is no gui. I went ahead and tried to build robocup but it errored saying that the file system is read-only.

jgkamat, 2016-07-28:
try using the file-> import appliance. Maybe it's designed to be imported, not used directly? There is a gui though.

chachmu, 2016-07-28:
@jgkamat Ok, I will give that a shot

chachmu, 2016-07-28:
@jgkamat That worked although you might want to increase your video memory some. My computer won't even let me make it fullscreen unless I triple it.

jgkamat, 2016-07-28:
Yah, I don't think you can set that before you import (all though I'll try). That will be one of the steps in my import guide if I cant do that. Bump it all the way up (128mb), as well as more ram and more cores.

Also disable the desktop server thing, that was causing issues for me (although Idk why its on).

chachmu, 2016-07-28:
@jgkamat If it can't be set beforehand then thats fine. Maybe also include instructions to add a disk drive to the vm so you can use virtualbox guestadditions? It's pretty helpful.

jgkamat, 2016-07-28:
Yes, this is a good idea! 

If you find any real issues running it (or actually anything tbh) just make issues on the images repo so I can track them. IMO this is a good start (at least).

jgkamat, 2016-08-21:
This seems like it's done, all tracking/issues can move to robocup-images.

closed 2016-08-21 by jgkamat