RoboJackets/robocup-software #781: Robots veer off course when turning and moving

jgkamat, 2016-07-16:
For an example, see this:
[image]

I'm not sure if this is a problem with the soccer PID or the per wheel PID, but I think its on the per wheel PID.

justbuchanan, 2016-07-16:
Tuning the pid in soccer should fix this.  There's no per-wheel pid in the simulator.

jgkamat, 2016-07-16:
So then the issue that we saw at comp was just improperly tuned PID, I'm guessing. Hopefully this isn't a bigger issue than that.

jgkamat, 2016-07-16:
I don't think this is an issue with the PID. 

I think the issue is that as we rotate at a higher speed, the rotation alters our sent commands (as our commands are sent in the robot's frame of reference), Since our rotation speed is increasing, we continue to accumulate errors even though the PID will try to compensate. I think we could fix this by predicting where the robot will face when the command is actually received (maybe length of a frame / 2) and using that angle instead of the current robot's angle to send velocities on. Do we do anything like this already?

ashaw596, 2016-07-16:
I tried that a bit and it didn't really help much (in Sim at least). I
think it's more a problem that in turning  cases the robot acts a bit more
like a car than the particle we model it as.

ashaw596, 2016-07-16:
Could also just be a Sim thing. We could test the robot itself by driving
forward and then switch to turning in place. Ideally it should slow down in
a straight line even if it's turning(though wheel momentum might be an a
issue). If it slows down while turning in a circle, it could be pid not
controlling the wheels well,  pid overcompensating, or our particle model
is weird with the wheels(would lqr help?). With our motors, the robot might
also just stop and fall over with really fast wheel deceleration.

jgkamat, 2016-11-14:
@jpfeltracco found out that this is an issue with us limiting our speed. We don't take rotation into account when driving, so when we rotate and drive, wheels at the top can't drive fast enough and we swerve.

jgkamat, 2017-01-15:
@jpfeltracco does #856 fix this? I thought it did, but I wanted to be sure.

jpfeltracco, 2017-01-16:
Yep, I'll go ahead and close this, I thought I mentioned in the PR that it closes this but I guess not.

closed 2017-01-16 by jpfeltracco