RoboJackets/robocup-software #782: CircleCI Style command no longer works

jgkamat, 2016-08-01:
This is because the curl url is now being redirected, we need to change the displayed `curl` command to actually resolve properly.

jgkamat, 2016-08-01:
changing `curl` to `wget -qO -` fixes this. I think everyone has wget installed, so we can probably just swap it out (unless someone figures out why curl isn't working with the redirect).

jgkamat, 2016-08-01:
On another note, the status tags might become impossible if circleci stops the redirects. We might want to look into a way of getting these URL's programatically.

jgkamat, 2016-08-03:
Style seems to be somewhat broken in general now too (checking more files than it needs to?)

I'll look into that as well.

justbuchanan, 2016-08-03:
Hmm, that's odd.  Do you think it's related to the new version of stylize that changed the `--diffbase` option?  I can look at it when I'm back at my computer later.

jgkamat, 2016-08-03:
I messed around with it and your diffbase seems to work (on my machine, in this scenario). I think that this is an option with the setup,

Don't look into it, I'll let you know if its an issue with stylize.

jgkamat, 2016-08-04:
@justbuchanan that particular issue was that I merged an old version of master, everything was working correctly on your end :smile: 

I also looked at the `merge-base` docs, and it's EXACTLY what we want! =)

commit d8f80287 (2016-08-07): Make curl to follow redirects

Fixes #782
closed 2016-08-07 by jgkamat