RoboJackets/robocup-software #787: CMake adding both -std=gnu++11 and std=c++14 when generating build.ninja.

jpfeltracco, 2016-08-29:
(This is only happening on my Arch system as far as I can tell.)

Building soccer was failing, complaining about std::make_unique calls changed in #306. Since these were added in c++14, it appeared that on my machine the build was using c++11.

I grepped through the repo and only found references in the build/build.ninja (not a source controlled file) generated by cmake which was specifying both -std=c++14 and -std=gnu++11:

`FLAGS = -std=c++14 -Werror=return-local-addr -g -Werror=return-type -Werror=delete-non-virtual-dtor -Werror=uninitialized -fPIC   -fPIC -std=gnu++11`

This doesn't happen in Ubuntu. I figured it might have been related to the CMake/Ninja version I was using (3.6.1 instead of 3.5.1 and 1.7.1 vs 1.5.1), however when manually installed 3.6.1/1.7.1 in Ubuntu, cleaned, and built, it worked fine and didn't add the extra -std=gnu++11 to the build.ninja file.

I fixed the issue on my arch machine by removing the CMAKE_CXX_FLAGS and replacing it by setting the CMAKE_CXX_STANDARD variable to 14 instead. I'm not sure if it's a more correct way to do it but it seems to work in both Ubuntu and Arch. The changes are in the fixed_up_build_arch branch right now.

jgkamat, 2016-08-29:
Its good that you got a chance to test out the arch build, I haven't had time to do it. It looks good to me, but I'm going to want to look a bit more into cmake docs before I :+1:. Thanks for doing the work though :smile: 

Could you open a PR from that branch into master, and we can continue the talk there?

jpfeltracco, 2016-08-29:
> more into cmake docs before I 👍. Thanks for doing the work though 😄

Yup, I'll do that first thing tomorrow!

jpfeltracco, 2016-08-31:
Here's the [related PR](https://github.com/RoboJackets/robocup-firmware/pull/20) in the firmware repo.

Both PR's should be good to go.

PR #789 (2016-08-31): Fixed up Arch build.
Added a missing sdl2 package and changed our method of specifying C++14.

I grabbed the CMakeLists snippet from [this website](https://crascit.com/2015/03/28/enabling-cxx11-in-cmake) which describes a few different approaches if you want to check those out.

closes  #787
closed 2016-08-31 by jgkamat