RoboJackets/robocup-software #796: Protobuf Versioning Warnings

jgkamat, 2016-09-22:
On a newer version of protobuf (I think protobuf3) I'm getting a ton of warnings when compiling:
```
[2/149] Running C++ protocol buffer compiler on /home/jay/Code/robocup-software/common/protobuf/Point.proto
[libprotobuf WARNING google/protobuf/compiler/parser.cc:547] No syntax specified for the proto file: Point.proto. Please use 'syntax = "proto2";' or 'syntax = "proto3";' to specify a syntax version. (Defaulted to proto2 syntax.)
[3/149] Running C++ protocol buffer compiler on /home/jay/Code/robocup-software/common/protobuf/Robot.proto
[libprotobuf WARNING google/protobuf/compiler/parser.cc:547] No syntax specified for the proto file: Robot.proto. Please use 'syntax = "proto2";' or 'syntax = "proto3";' to specify a syntax version. (Defaulted to proto2 syntax.)
[libprotobuf WARNING google/protobuf/compiler/parser.cc:547] No syntax specified for the proto file: Control.proto. Please use 'syntax = "proto2";' or 'syntax = "proto3";' to specify a syntax version. (Defaulted to proto2 syntax.)
[4/149] Running C++ protocol buffer compiler on /home/jay/Code/robocup-software/common/protobuf/Control.proto
[libprotobuf WARNING google/protobuf/compiler/parser.cc:547] No syntax specified for the proto file: Control.proto. Please use 'syntax = "proto2";' or 'syntax = "proto3";' to specify a syntax version. (Defaulted to proto2 syntax.)
[5/149] Running C++ protocol buffer compiler on /home/jay/Code/robocup-software/common/protobuf/RadioTx.proto
[libprotobuf WARNING google/protobuf/compiler/parser.cc:547] No syntax specified for the proto file: Robot.proto. Please use 'syntax = "proto2";' or 'syntax = "proto3";' to specify a syntax version. (Defaulted to proto2 syntax.)
[libprotobuf WARNING google/protobuf/compiler/parser.cc:547] No syntax specified for the proto file: Control.proto. Please use 'syntax = "proto2";' or 'syntax = "proto3";' to specify a syntax version. (Defaulted to proto2 syntax.)
[6/149] Running C++ protocol buffer compiler on /home/jay/Code/robocup-software/common/protobuf/RadioRx.proto
[libprotobuf WARNING google/protobuf/compiler/parser.cc:547] No syntax specified for the proto file: RadioRx.proto. Please use 'syntax = "proto2";' or 'syntax = "proto3";' to specify a syntax version. (Defaulted to proto2 syntax.)
[7/149] Running C++ protocol buffer compiler on /home/jay/Code/robocup-software/common/protobuf/SimCommand.proto
[libprotobuf WARNING google/protobuf/compiler/parser.cc:547] No syntax specified for the proto file: SimCommand.proto. Please use 'syntax = "proto2";' or 'syntax = "proto3";' to specify a syntax version. (Defaulted to proto2 syntax.)
[libprotobuf WARNING google/protobuf/compiler/parser.cc:547] No syntax specified for the proto file: Point.proto. Please use 'syntax = "proto2";' or 'syntax = "proto3";' to specify a syntax version. (Defaulted to proto2 syntax.)
[8/149] Running C++ protocol buffer compiler on /home/jay/Code/robocup-software/common/protobuf/messages_robocup_ssl_detection.proto
[libprotobuf WARNING google/protobuf/compiler/parser.cc:547] No syntax specified for the proto file: messages_robocup_ssl_detection.proto. Please use 'syntax = "proto2";' or 'syntax = "proto3";' to specify a syntax version. (Defaulted to proto2 syntax.)
[9/149] Running C++ protocol buffer compiler on /home/jay/Code/robocup-software/common/protobuf/messages_robocup_ssl_wrapper.proto
[libprotobuf WARNING google/protobuf/compiler/parser.cc:547] No syntax specified for the proto file: messages_robocup_ssl_wrapper.proto. Please use 'syntax = "proto2";' or 'syntax = "proto3";' to specify a syntax version. (Defaulted to proto2 syntax.)
[libprotobuf WARNING google/protobuf/compiler/parser.cc:547] No syntax specified for the proto file: messages_robocup_ssl_detection.proto. Please use 'syntax = "proto2";' or 'syntax = "proto3";' to specify a syntax version. (Defaulted to proto2 syntax.)
[libprotobuf WARNING google/protobuf/compiler/parser.cc:547] No syntax specified for the proto file: messages_robocup_ssl_geometry.proto. Please use 'syntax = "proto2";' or 'syntax = "proto3";' to specify a syntax version. (Defaulted to proto2 syntax.)
[10/149] Running C++ protocol buffer compiler on /home/jay/Code/robocup-software/common/protobuf/messages_robocup_ssl_refbox_log.proto
[libprotobuf WARNING google/protobuf/compiler/parser.cc:547] No syntax specified for the proto file: messages_robocup_ssl_refbox_log.proto. Please use 'syntax = "proto2";' or 'syntax = "proto3";' to specify a syntax version. (Defaulted to proto2 syntax.)
[libprotobuf WARNING google/protobuf/compiler/parser.cc:547] No syntax specified for the proto file: messages_robocup_ssl_detection.proto. Please use 'syntax = "proto2";' or 'syntax = "proto3";' to specify a syntax version. (Defaulted to proto2 syntax.)
[11/149] Running C++ protocol buffer compiler on /home/jay/Code/robocup-software/common/protobuf/referee.proto
[libprotobuf WARNING google/protobuf/compiler/parser.cc:547] No syntax specified for the proto file: referee.proto. Please use 'syntax = "proto2";' or 'syntax = "proto3";' to specify a syntax version. (Defaulted to proto2 syntax.)
[12/149] Automatic moc for target soccer
Generating moc compilation soccer_automoc.cpp
[13/149] Running C++ protocol buffer compiler on /home/jay/Code/robocup-software/common/protobuf/messages_robocup_ssl_geometry.proto
[libprotobuf WARNING google/protobuf/compiler/parser.cc:547] No syntax specified for the proto file: messages_robocup_ssl_geometry.proto. Please use 'syntax = "proto2";' or 'syntax = "proto3";' to specify a syntax version. (Defaulted to proto2 syntax.)
[14/149] Running C++ protocol buffer compiler on /home/jay/Code/robocup-software/common/protobuf/LogFrame.proto
[libprotobuf WARNING google/protobuf/compiler/parser.cc:547] No syntax specified for the proto file: LogFrame.proto. Please use 'syntax = "proto2";' or 'syntax = "proto3";' to specify a syntax version. (Defaulted to proto2 syntax.)
[libprotobuf WARNING google/protobuf/compiler/parser.cc:547] No syntax specified for the proto file: referee.proto. Please use 'syntax = "proto2";' or 'syntax = "proto3";' to specify a syntax version. (Defaulted to proto2 syntax.)
[libprotobuf WARNING google/protobuf/compiler/parser.cc:547] No syntax specified for the proto file: Point.proto. Please use 'syntax = "proto2";' or 'syntax = "proto3";' to specify a syntax version. (Defaulted to proto2 syntax.)
[libprotobuf WARNING google/protobuf/compiler/parser.cc:547] No syntax specified for the proto file: messages_robocup_ssl_wrapper.proto. Please use 'syntax = "proto2";' or 'syntax = "proto3";' to specify a syntax version. (Defaulted to proto2 syntax.)
[libprotobuf WARNING google/protobuf/compiler/parser.cc:547] No syntax specified for the proto file: messages_robocup_ssl_detection.proto. Please use 'syntax = "proto2";' or 'syntax = "proto3";' to specify a syntax version. (Defaulted to proto2 syntax.)
[libprotobuf WARNING google/protobuf/compiler/parser.cc:547] No syntax specified for the proto file: messages_robocup_ssl_geometry.proto. Please use 'syntax = "proto2";' or 'syntax = "proto3";' to specify a syntax version. (Defaulted to proto2 syntax.)
[libprotobuf WARNING google/protobuf/compiler/parser.cc:547] No syntax specified for the proto file: Robot.proto. Please use 'syntax = "proto2";' or 'syntax = "proto3";' to specify a syntax version. (Defaulted to proto2 syntax.)
[libprotobuf WARNING google/protobuf/compiler/parser.cc:547] No syntax specified for the proto file: Control.proto. Please use 'syntax = "proto2";' or 'syntax = "proto3";' to specify a syntax version. (Defaulted to proto2 syntax.)
[libprotobuf WARNING google/protobuf/compiler/parser.cc:547] No syntax specified for the proto file: RadioRx.proto. Please use 'syntax = "proto2";' or 'syntax = "proto3";' to specify a syntax version. (Defaulted to proto2 syntax.)
```

Probably it's an easy fix with the protobuf files. I think 16.04 isn't affected yet. This does not effect compilation.

jgkamat, 2016-09-22:
This is a possible candidate for an intro issue, moving to the intro project.

ashaw596, 2016-09-22:
Um. Proto 2 and 3 is more like python 2 and 3.....

ashaw596, 2016-09-22:
Though with slightly better versioning for backwards compatibility.

jgkamat, 2016-09-22:
I would rather keep the files on proto2 for now (as 14.04 probably dosen't have proto3). All that needs to be done for this issue is to squelch the warnings.

jgkamat, 2016-10-02:
This issue can be solved by adding:
```
syntax="proto2";
```

to the top of every file in [this folder (in the common repo)](https://github.com/RoboJackets/robocup-common/tree/master/protobuf)

You'll either need to know a bit about how to commit inside a submodule or just clone the common repo seperately to work on this.

mszylkowski, 2016-11-29:
I'll take it

closed 2016-12-11 by jgkamat

jgkamat, 2017-03-29:
For posterity, this was fixed in robojackets/robocup-common@dfe6d9c1eaff9621d13495abacbaff32c299f81f

ahao1011, 2017-06-28:
wo have add syntax="proto2";  but Error is still there

AHdeMacBook-Pro:Gen ah$ protoc --proto_path=. --objc_out=. Person.proto
[libprotobuf WARNING google/protobuf/compiler/parser.cc:546] No syntax specified for the proto file: Person.proto. Please use 'syntax = "proto2";' or 'syntax = "proto3";' to specify a syntax version. (Defaulted to proto2 syntax.)
Person.proto:1:1: Expected top-level statement (e.g. "message").
Person.proto:4:1: Expected top-level statement (e.g. "message").
Person.proto:5:1: Expected top-level statement (e.g. "message").
Person.proto:33:3: Expected top-level statement (e.g. "message").
Person.proto:34:1: Unmatched "}".

ashaw596, 2017-06-28:
What Person.proto file is this? Is it in the repo? It seems to still not see the syntax version statement. You should verify that it is at the top and exactly correct. Maybe copy and paste it from the error message.

jgkamat, 2017-06-28:
@ahao1011 
By that error message it seems like you have some other errors as well, which might be confusing the protobuf compiler. :+1: on the syntax directive being at the top of the file.

If you give us a link to the protobuf file, I might be able to help you fix the problem. Let us know if it's a file in any of `RoboJackets/*` and I'll go ahead and fix that.

ahao1011, 2017-06-28:
@jgkamat  https://github.com/ahao1011/Gen  the protobuf file link , this is my first time to try make a proto.  thank you

jgkamat, 2017-07-10:
@ahao1011 sorry for the delay on this, I've been kind of busy.
Adding `syntax="proto2";` to the top of the file removes the warning about syntax versioning, but there are a lot of other unrelated errors.

[This is the protobuf file that has the syntax warning removed](http://paste.debian.net/975670/). However, you still have a lot of other errors:

```
Person.proto:2:1: Expected top-level statement (e.g. "message").
Person.proto:6:1: Expected top-level statement (e.g. "message").
Person.proto:7:1: Expected top-level statement (e.g. "message").
Person.proto:10:40: Invalid escape sequence in string literal.
Person.proto:10:40: String literals cannot cross line boundaries.
Person.proto:11:14: Expected top-level statement (e.g. "message").
Person.proto:37:3: Expected top-level statement (e.g. "message").
Person.proto:38:1: Unmatched "}".
```
You have a lot of weird characters in your protobuf file, so I would suggest starting again from scratch and trying again. 

You can look at any of our [existing protobuf](https://github.com/RoboJackets/robocup-common/tree/master/protobuf) files for examples, if that helps.

I've tried to clean up your protobuf file by cleaning out everything that I didn't understand, which resulted in [this protobuf file](http://paste.debian.net/975672/), which compiles fine. 

If you need additional help, feel free to open an issue on your repo, and tag me, but we should try to limit conversation on closed issues, so things stay organized. :smile:

PR #842 (2016-12-11): Update common to master
Closes #796.

Needs a bit of testing, I'll manage that.