RoboJackets/robocup-software #804: Create Dark Theme for Soccer

jgkamat, 2016-10-02:
I really like dark themes. Bonus points if you use my dark theme.

jgkamat, 2016-10-03:
To do this, a theme switching interface/mecahnism must be made as well, as we want to keep the current theme as a default.

jgkamat, 2016-11-28:
This can be done by changing the [stylesheet](https://github.com/EClaesson/QTDark/blob/master/QTDark.stylesheet) that qt uses to display our app.

jgkamat, 2017-01-13:
Just to help aggregate more resources:

This might be an interesting method (although you will need the stylesheet as a string):
http://doc.qt.io/qt-5/qwidget.html#styleSheet-prop

These are docs on QT Style sheets in general: http://doc.qt.io/qt-5/stylesheet.html

In addition to `styleSheets`, there are also qt `styles` (which change the appearance of buttons, but not their color, as far as I know. This method can be used to set qt Styles: http://doc.qt.io/qt-5/qapplication.html#setStyle

The actual object to execute these functions on _should_ be this one: https://github.com/RoboJackets/robocup-software/blob/master/soccer/main.cpp#L61

jgkamat, 2017-01-15:
Also, a nice to have would be a file that soccer would read that would set the default stylesheet.

Maybe `~/.soccerrc`? I can probably implement this after this is completed, but once this is done, I would love to have the stylesheet set every time I launch soccer without changing it manually.

jgkamat, 2017-01-15:
Actually on second thought, a config file (like `~/.soccerrc`) would be easier to implement (no messing around with UI's) and more practical (no one wants to set the styleshset every time), so if you wanted, you could implemente stylesheet to be set exclusively with a file like that.

Such a system needs to be thought out properly, so the same file could be used for other user specific changes. We already have a config system for overall configuration, but this probably dosen't belong there, since that file is version controlled (and user's can't change it without git annoying them about the file being changed).

Jason27chan, 2017-01-16:
Please help. a Style Sheet Manager, but it gives me an undefined function
Here's my line of code
`StyleSheetManager manager(this);`
Here's the error
`../run/robocup.so: undefined reference to `StyleSheetManager::StyleSheetManager(MainWindow*)'`

chachmu, 2017-01-16:
Does StyleSheetManager exist and/or have you included it in the file?

Jason27chan, 2017-01-16:
Yeah, here's the line where I included it
`#include <ui/StyleSheetManager.hpp>`

jgkamat, 2017-01-16:
Hmm, I can't tell for sure without seeing the code, but maybe you haven't defined a constructor to StyleSheetManager that takes in a MainWindow. Usually when you get linker errors (like this one) it means you have the prototype (the hpp file) but you don't have the implementation in the cpp file.

If you've defined a new cpp file (I think you have), you will also need to add it to this file: https://github.com/RoboJackets/robocup-software/blob/master/soccer/CMakeLists.txt#L28

That way it actually gets compiled in the build process. Let me know what happens!

Jason27chan, 2017-01-18:
How would you set a global environment variable in bashrc? I just tried doing var=DARK, and I still need to export the variable for it to be used

jgkamat, 2017-01-18:
Talked about this in the meeting, but for everyone else:

```
export var=DARK
```
is what you need for env vars to be seen by child programs.

Jason27chan, 2017-01-20:
I got switching themes in the program to work and I got the simulator to access the environmental variable. The problem is that I can't run them simultaneously because I use QMainWindow.setStyleSheet for the former and QApplication.setStyleSheet for the latter, and the latter overwrites the former

jgkamat, 2017-01-21:
Hmm, I don't think we need stylesheets on the simulator, just soccer (since we don't use the simulator in non-headless mode very much). 

I'm not exactly sure what QApplication.setStyleSheet does (as opposed to QMainWindow), could you send some pictures of both behavior (so I can tell the difference?)

PR #884 (2017-02-06): Final Tweaks and release of Themes
Continuation of #880

Planning to clean up the UI buttons to be radio boxes, and add documentation.

Closes #804
Thanks to @Jason27chan for all the hard work! :smile:
closed 2017-02-06 by jgkamat