RoboJackets/robocup-software #809: Can't bind to shared vision port

jgkamat, 2016-10-30:
Starting soccer after vision causes this error: `Can't bind to shared vision port`. 

This is caused because of:

1. vision binding the port
2. https://bugreports.qt.io/browse/QTBUG-33419

jgkamat, 2016-10-30:
Since that bug report has basically no activity, the easiest solution would be to switch to a different UDP library.

jgkamat, 2016-11-01:
The easiest way to get around this  (I think) would be to switch our udp packet receiver to use the boost library rather than QT. I (hope) the boost library is better maintained and would work well.

[here is an example of this](http://www.boost.org/doc/libs/1_37_0/doc/html/boost_asio/example/multicast/receiver.cpp). Shouldn't be too hard, between our current implementation and the example.

jpfeltracco, 2016-11-01:
We should try adding the "QUdpSocket::ReuseAddressHint" flag as well to soccer's bind. Docs are here http://doc.qt.io/qt-4.8/qudpsocket.html

jgkamat, 2016-11-01:
Yah we tried that already, with no effect. That flag only helps on windows, for unixes, the ShareAddress flag we have should do everything (and did work properly before).: https://doc.qt.io/qt-4.8/qudpsocket.html#BindFlag-enum

jpfeltracco, 2016-11-01:
Ah okay, that's weird!

jgkamat, 2016-11-04:
I think qt 5.6 solves this issue, but 16.04 is on qt 5.5. *Sigh*

[image]

jpfeltracco, 2016-11-04:
Time to add all the PPAs!

jgkamat, 2017-02-04:
We patched ssl-vision to work around this (ssl-vision really shouldn't be binding to the vision port, but it does...). There's not much we can do (dammit QT), so I'm closing this.

Hoepfully it will come up in people's searches if they run into this.

closed 2017-02-04 by jgkamat