RoboJackets/robocup-software #825: 'Red Cards' label in UI stylesheet is off

jgkamat, 2016-11-27:
It looks kind of wrong right now:

[image]

You could fix this in qtcreator, but it's honestly easier to edit this file by hand instead. 

[This is the bad section of the ui file](https://github.com/RoboJackets/robocup-software/blob/6ea6cf696dd5ecd7caca044bee93a308cea32e5c/soccer/ui/MainWindow.ui#L1271)

[Here is a good section from the blue team's 'red card' section:](https://github.com/RoboJackets/robocup-software/blob/6ea6cf696dd5ecd7caca044bee93a308cea32e5c/soccer/ui/MainWindow.ui#L1113).

Should be enough to replace the broken one with the ok one, and readjust the colors to match yellow.

krispreza, 2016-11-29:
Ill take this.

EdwardTan20, 2016-11-29:
so this file will be rewritten or replaced?

closed 2016-11-30 by jgkamat