RoboJackets/robocup-software #828: Python complaining about double initialized methods

jgkamat, 2016-11-27:
```
_frozen_importlib:222: RuntimeWarning: to-Python converter for OurRobot* already registered; second conversion method ignored.
_frozen_importlib:222: RuntimeWarning: to-Python converter for OpponentRobot* already registered; second conversion method ignored.
_frozen_importlib:222: RuntimeWarning: to-Python converter for SystemState* already registered; second conversion method ignored.
_frozen_importlib:222: RuntimeWarning: to-Python converter for std::shared_ptr<Configuration> already registered; second conversion method ignored.
_frozen_importlib:222: RuntimeWarning: to-Python converter for ConfigBool* already registered; second conversion method ignored.
_frozen_importlib:222: RuntimeWarning: to-Python converter for ConfigDouble* already registered; second conversion method ignored.
_frozen_importlib:222: RuntimeWarning: to-Python converter for ConfigInt* already registered; second conversion method ignored.
```
appears when running soccer.

This can be solved by [removing all instances of register_ptr_to_python in this file](https://github.com/RoboJackets/robocup-software/blob/4da80ab54496394212a0a618ff58ad10642a92cf/soccer/gameplay/robocup-py.cpp#L655). 

Right now I can't see any problems when we do this, but whoever's trying this should try and see if they can break any python stuff we have.

As far as I know these do nothing (hence the warning) as of now, so removing them shouldn't be an issue.

Jason27chan, 2016-11-29:
got it

jcarn, 2017-02-08:
Was this fixed? It still happens for me

ashaw596, 2017-02-08:
Oh I undid the fix since it broke it on my mac.

jgkamat, 2017-02-08:
Yah, we might want to figure out why mac needs this but ubuntu complains about it. I'm not sure what approach to take on this :/

JNeiger, 2019-01-18:
Looks like the changes got reverted to Jason's branch. Imma close this.

PR #840 (2016-12-04): Fixed python complaining about double initialized errors
Closes #828
closed 2019-01-18 by JNeiger