RoboJackets/robocup-software #830: Repeated line up shoud provide an easy way to traverse only part of the field

jgkamat, 2016-11-28:
Repeated line up currently traverses the entire field, whcih is what we want by default.

However, during tuning, we may want to insert a buffer, to limit how far we command the robots to go, so if there is a problem, we won't bump into the walls.

[This should be easily achevable by changing the calls to gen_line here to decimal values](https://github.com/RoboJackets/robocup-software/blob/8aa13ff4334af3d8bba1f5e64febd36c311c8d38/soccer/gameplay/plays/testing/repeated_line_up.py#L54). However, it would be nice to have a single constant value at the top of the file to tune both of these gen line calls at once, so it's easy to do this in a rush.

ashaw596, 2016-11-28:
Relevant idea which I just remembered to create an issue for #832

rk1998, 2016-11-29:
ill take this

PR #838 (2016-12-04): Created global variable to limit how far robot's travel
I created a variable called BUFFER that can be set to any value and it will change how far the robot moves across the field during repeated lineup. BUFFER should be a decimal value.

Closes #830
closed 2016-12-04 by jgkamat