RoboJackets/robocup-software #832: Create Configurable Points which can be used in testings and config plays

ashaw596, 2016-11-28:
Right now we have configurable constants which we use. This idea is to be able to set "Configurable points" on the config tab which basically contain X and Y poisitions, but can be set by clicking on the field. This could be generalizalble like our config system or just create like 4-8 set points like ConfigPoint1, ConfigPoint2, ConfigPoint3, and ConfigPoint4 which the plays can choose to use. 

Examples of use case is have an easily changed point that the play just moves a robot to. Points that can be set as the endpoints for lineup. It can take in a couple points like 4 and use them at waypoints to test pathplanning. We can set waypoints to test passing with different distances.

justbuchanan, 2016-11-28:
Related to #425

ashaw596, 2016-11-28:
Oh right! I could have sworn we had an issue, but I couldn't find it. That
ones better.

jgkamat, 2016-11-28:
I really like this idea, although I don't know how hard it would be to implement (and whether it would be worth it).

ashaw596, 2016-12-02:
Closed since pretty much duplicate of #425

closed 2016-12-02 by ashaw596