RoboJackets/robocup-software #859: Quickload/Quicksave shortcuts don't work on ubuntu

chachmu, 2017-01-15:
This may just be an issue on my vm but it would be nice if someone else would check. I have seen Albert use the shortcuts (Q and E) on his mac so I don't think its a problem there.

jgkamat, 2017-01-15:
Huh, thats weird, they definetly work for me as well. 

Definetly a good begginer issue, might need a bunch of debugging though.

chachmu, 2017-01-15:
If its just me it's not that big of a deal, I just didn't have a way to tell if it was for everyone on not. I'm fine with closing this if you want to.

jgkamat, 2017-01-15:
Yah, if you're running the ubuntu VM we definetly want to fix this. If it's affecting you its probably affecting everyone else as well (and we should probably figure out the root cause anyway).

chachmu, 2017-01-15:
I'm not sure exactly what might cause this but we should definitely see if it happens on the premade image. I am not using that so it might be an issue with how I installed qt or soccer itself

petschekr, 2017-01-18:
Confirming this bug on a native Ubuntu 16.04 LTS installation. Saving / loading from the `Simulator` menu works fine but the keyboard shortcuts do not.

petschekr, 2017-01-18:
Seems to be caused by [this bug in appmenu-qt5](https://bugs.launchpad.net/appmenu-qt5/+bug/1380702) which affects Ubuntu. Changing `./run/soccer -sim -pbk example.pbk` in the makefile to `env UBUNTU_MENUPROXY= ./run/soccer -sim -pbk example.pbk` fixes this issue but alters the way that the top toolbar is drawn. However, this workaround also prevents the integration of the toolbar items into Ubuntu's GUI and draws the toolbar at the top of the window instead of at the top of the screen.

jgkamat, 2017-01-18:
O.O

Dammit ubuntu.

I'm not sure if there's anything we can do about this. The env variable should probably be set in user's bashrc files instead of in our makefile, but I'm not sure. 

Thoughts?

jgkamat, 2017-04-05:
It looks [like ubuntu just announced they are moving to gnome](https://insights.ubuntu.com/2017/04/05/growing-ubuntu-for-cloud-and-iot-rather-than-phone-and-convergence/), and this issue will likeley be fixed by that (when 18.04 ships).

jgkamat, 2017-04-22:
Since this is an upstream issue that we can't do anything about (other than report to ubuntu/qt), I'm closing this.

closed 2017-04-22 by jgkamat