RoboJackets/robocup-software #861: Fix RRT Planning Weirdness

jpfeltracco, 2017-01-16:
I talked to @joshhting about this a bit. The path planner seems to be choosing some insane paths from time to time when running the repeated turning lineup play. It doesn't happen consistently, just every once in a while. See the images below for examples.

[image]

[image]

jgkamat, 2017-01-16:
Woah, I've never seen this before. Have you seen this only on your computer? 

This might be happening because:

1. We can't go through the goal zone
2. The 2nd lowest robot might be too low to allow the lowest one pass
3. the ball makes the path go even farther up.

So there might not be an explicit bug, but I'm not sure. There should be an option to turn on RRT debug visualization which would help in figuring this out a lot!

joshhting, 2017-01-16:
I believe the rrt implementation that we use in robocup-software currently has no waypoint or goal bias. If we increase the goal bias, this likely wouldn't be a problem anymore.

It's also weird that in the first image, the extremely absurd path somewhat intersects other robots. Do we not consider our own robots to be obstacles? Or if we do, maybe the obstacle boundaries are too small?

jgkamat, 2017-01-16:
We do, but there's a time factor as well. We can have paths crossing each other, if the robots will be in the same place at different times.

ashaw596, 2017-01-17:
This looks similar to the problems I was seeing when we were merging the new RRT (abeit more extreme).
https://www.dropbox.com/s/nhtkrdz7j1z1wha/Screenshot%202016-11-05%2003.31.01.png?dl=0

We weren't having much luck debugging it before, but we should be able to figure it out by looking at the RRT searches. These paths really shouldn't be possible with optimization. If its a bug with collisions we could try turning those off and see if these still get generated. Don't really think waypoint biases and stuff would fix it. Might be issues with those too, but the optimizer should shorten the paths.

ashaw596, 2017-01-17:
Also, the original RRT Pathplanner always tried a direct path first. This might not the the case anymore.

jpfeltracco, 2017-01-18:
Here's a picture of the planning. The issue is only with the initial paths they plan, and are definitely related to the goal being an obstacle. I bumped the y position of the play slightly and didn't really see any more issues.

[image]

joshhting, 2017-01-18:
Shouldn't it still smooth the paths out into a straight line afterwards when it constructs the Bezier curve?

justbuchanan, 2017-01-18:
I just opened #873 which I think helps this some, although it doesn't completely fix it. Try it out and see if you can get any awful paths with those changes.

I can get similar results to the screenshot two messages above, but nothing quite as bad as the first screenshots in the thread. Can you reproduce the really bad ones with the rrt visualizations on?

Some of the crazier paths might be because of collisions with other paths we've generated, resulted in "dynamic obstacles" being added before another iteration is run. I add code to draw dynamic obstacles in [this branch](https://github.com/RoboJackets/robocup-software/tree/justin/rrt-draw-dynamic-obstacles), which lets you see when this is happening. It looks like that doesn't actually happen all that often under normal circumstances.

justbuchanan, 2017-01-18:
I just realized this issue was about the *turning* repeated line up play... My changes may not do much for this if the issue is related to the angle path stuff.

I've played with this a little more (on the min-iterations branch) and I can still get some bad paths, but a lot of them look like they're just bad luck. In the screenshot below for robot 5, it explored a good number of iterations, but because the window it had to go through for the shorter path was too small, it never got it. It might help to play around with the avoid radius with our own robots - they're not drawn on the screen, but I think itls ~1.5x the actual robot radius. This case could also be improved by increasing minIterations so it searches for longer, but that has the obvious tradeoff that it makes our planning even more cpu-intensive than it already is.

[image]

jpfeltracco, 2017-01-22:
I haven't been able to recreate any paths as bad as the first ones, maybe the RRT was just having a bad day.

I also agree that it appears that the bad paths on the lowest robot are just due to the lack of space between the goal and the robot above it. I think the only reasonable path it finds is fairly high up, then when the robots near it follow their paths a bit and clear the area, the lowest robot is able to find a good path.

So maybe this is just a non-issue and the best we can do is adding that minIterations to ensure we have decent paths.

jgkamat, 2017-07-03:
I haven't been able to repro this for a long time, so I'm closing this. We'll reopen if we see it again.

closed 2017-07-03 by jgkamat