RoboJackets/robocup-software #863: Goalie Leaves the Goalie Zone when clearing a ball

jgkamat, 2017-01-16:
The normal robot rules (which are to stay out of the goalie zone) are being applied when the goalie enters the clearing state (once the ball enters the goalie zone). This is pretty bad, since the goalie runs away from the ball, leaving the goal open.

This might be a good beginner issue, but I'm not sure. Its very important though.

jgkamat, 2017-01-16:
This is reproducable in the simulator, so you don't need actual robots to tackle this issue.

zhangj150, 2017-01-25:
@jgkamat this should be demoted in importance.

jgkamat, 2017-01-25:
@zhangj150 you should be invited now! Make sure to assign yourself to the issue you decide to work on (we discussed #591 in the meeting today)

jgkamat, 2017-01-25:
Also this issue seems to only be reproducable in the stopped state (not in actual play) so this isn't as important as I thought.

Jason27chan, 2017-08-29:
@jgkamat Can't seem to reproduce this issue. Is it fixed?

jgkamat, 2017-08-30:
Yup, thanks for the bump. I think I fixed this at some point, I can't remember when.

closed 2017-08-30 by jgkamat