RoboJackets/robocup-software #866: When stopped, halting and re-stopping play causes robots to shift positions

jgkamat, 2017-01-16:
This seems kind of weird, and might be a red flag of some bigger problems in our role assignments.

When we stop play, halt, and restart play, no robots should move, because everything is in their correct places. However, robots often switch places for no apparent reason. This may be due to the ball obstacle for some reason, but this needs to be triaged to figure out the underlying cause.

ashaw596, 2017-01-17:
Um, this might be an issue with our role requirements stuff too. This might be rather difficult.

jcarn, 2017-02-15:
I'll check this out

jcarn, 2017-03-01:
I found a way to do it but it is pretty hacky, basically forcing the stopped play's reqs to use the same robot they just used. This has some obvious issues, and I think the real issue is in how role_reqs is assigning cost to the reqs, but if we need a quick fix it exists.

jgkamat, 2017-03-01:
I'd rather find the root cause, I'm worried that this is a symptom of a larger issue. This particular problem isn't too bad, so I don't think that it's worth it to add a specific workaround.

jcarn, 2017-03-06:
The issue looks like role requirements are being given incorrect shell_ids. I found where this happens, and removing the line fixes the issue and doesn't seem like it affects anything else, but I want to talk to you about what might be going on.

jgkamat, 2017-03-06:
Sounds like a plan, I'm glad you have a lead though! :tada:

PR #920 (2017-04-05): Fix rotation bug when halting during stop play
Closes #866 . So the following was happening:

1. Stop play starts and selects a tactic (circle around ball vs. circle around center)
2. Tactic doesn't know how many robots it will receive so it creates 6 move behaviors with destinations evenly spaced around the circle
3. Role assignment occurs and the tactic might not receive 6 robots, usually it gets 3
4. The (usually 3) robots are assigned to 3 of the 6 points around the circle based on their proximity to those original points
5. The move behaviors' points are updated now that the tactic knows  [... cut]
closed 2017-04-05 by jgkamat