RoboJackets/robocup-software #871: Goalie should intercept ball when it is shot at goal

krispreza, 2017-01-18:
Right now the goalie moves according to the position of the ball on the field, but when the ball is shot at the goal we want the goalie to move to cut off the angle of the shot.

krispreza, 2017-01-18:
There is code in goalie.py that will move the goalie preemptively in order to block a shot however this code only runs when an opponent robot has the ball and is preparing to take the shot. This might solve the issue of the goalie not intercepting the ball in time. However I am not sure how to test this in soccer.

jgkamat, 2017-01-18:
We might want to change this behavior to do this transition on any shot, even if we just kick it without the opponent robots. I'm not sure if this would be a performance problem though.

jgkamat, 2017-02-08:
I think this was only the case when running the stopped play, we have to re-evaluate this issue.

jgkamat, 2017-04-22:
On second thought, not intercepting the ball is probably the correct action to do (to avoid accidents on the field). Stopped should do the minimum amount of work possible. grSim and optimizations show that (at least in simulator), we do try to block the ball.

closed 2017-04-22 by jgkamat