RoboJackets/robocup-software #878: Drawing in Python fails

chachmu, 2017-01-22:
The system_state().draw_circle(... and other functions seem to not be properly drawing. I git bisected while testing to see if coordinated pass drew the triangle when passing and found that it stopped working here: 8e270b40ca13eef359f15122eb13703a2175463a although I can't immediately see anything wrong with the commit.

JNeiger, 2017-01-22:
I found that you have to use the predefined names in the layers tab within the simulator. I think theres "Debug", "Info[1-4]" and a few others. 

See [here ](https://github.com/JNeiger/robocup-software/blob/adaptive_formation/soccer/gameplay/plays/testing/test_adaptive_formation_weights.py#L94)for an example

chachmu, 2017-01-22:
The problem is that I don't think any of the code reflects that. As far as I can tell coordinated pass and Placement have always used that string as a description of what is being drawn. I don't know which is better but if we decide to stick with requiring the string to describe the layer we would need to change all of the places that the draw system is used incorrectly.

JNeiger, 2017-01-23:
I think I found the fix. Good find on that commit, that helped a bunch

PR #881 (2017-01-23): Fixes Python Draw Functions
Closes #878 

Before, only a copy of the global `state` object was passed so it would be reset to the default by the next frame. This should allow the changes to the layers list to be propagated back up to the main `state` variable and allow it to exist across frames.
closed 2017-01-23 by JNeiger