RoboJackets/robocup-software #883: Make behavior-diagrams throws a bunch of errors

Jason27chan, 2017-01-25:


jgkamat, 2017-01-25:
An example of the errors being thrown:

```
ERROR:root:Error generating fsm diagram for behavior 'Basic122':'NoneType' object has no attribute 'isChecked'
Traceback (most recent call last):
  File "generate_fsm_diagrams.py", line 37, in <module>
    klass().write_diagram_png(filepath)
  File "/home/jay/Code/robocup-software/soccer/gameplay/plays/offense/basic_122.py", line 27, in __init__
    super().__init__(continuous=False)  # FIXME: continuous?
  File "/home/jay/Code/robocup-software/soccer/gameplay/standard_play.py", line 17, in __init__
    self.use_standard_defense()
  File "/home/jay/Code/robocup-software/soccer/gameplay/standard_play.py", line 24, in use_standard_defense
    if ui.main.defenseEnabled() and not self.has_subbehavior_with_name(
  File "/home/jay/Code/robocup-software/soccer/gameplay/ui/main.py", line 24, in defenseEnabled
    return _defense_checkbox.isChecked()
AttributeError: 'NoneType' object has no attribute 'isChecked'
```

jgkamat, 2017-03-16:
Fixed in #895

closed 2017-03-16 by jgkamat