RoboJackets/robocup-software #885: Tweak Colors in Robot Status Panel

guyfleeman, 2017-01-29:
The robot number is currently gray when everything is connected and working properly. It should probably be gray when disconnected, red when there are errors, and green when everything is okay.

Maybe yellow for radio but no vision as well?

EdwardTan20, 2017-02-12:
ill work on this

hchittanuru3, 2017-02-22:
To confirm, should the Robot Status Widget show orange even for hall faults, as it states in issue #418?

chachmu, 2017-02-22:
I think that issue was referring to the "tab" in the section on the right, which is currently not relevant because we haven't moved the robot status window yet.

jgkamat, 2017-02-22:
@hchittanuru3 I think that hall faults should be orange (the colors should probably match whatever's in the status panel, when we decide to move it into the tab widget). I don't think we're ready to move it into there just yet :smile:.

I would try to make it as easy as possible to change the colors, so we can extend/change colors easily, so I wouldn't worry too much about the colors, we can decide the exact behavior once we have a basic protoype.

closed 2025-09-04 by N8BWert