RoboJackets/robocup-software #892: Switching from Win_Eval to a new method

JNeiger, 2017-02-03:
I think I've found a better way to calculate whether we are clear for a shot / pass than win_eval. Additionally, it has a few more options to change sensitivity in comparison and we can deal with the case of a moving robot much more easily. The only difference is that this would only return the chance, it would not return the most optimal windowed area.

At least in the matlab implementation of each, the new method preforms significantly faster. In the order of .75 seconds while the win_eval takes about 34 seconds for all ~5000 points in the graph. If this speed difference transfers over to the soccer, I think the added complexity of two different functions might be worthwhile. 

In the image below, the dark parts in the "Space" graph are where the robots are. Yellow is 100% while blue is 0%. The pass target is the bottom center and the color at each point represent the pass chance.

[image]

Thoughts?

[Here is a link](https://github.com/JNeiger/robocup-software/blob/adaptive_formation/soccer/gameplay/evaluation/defensive_positioning.py#L78) to the current iteration. It's a part of #822 right now. I'll get around to cleaning it up eventually.

ashaw596, 2017-02-03:
@barulicm

barulicm, 2017-02-03:
@JNeiger 
The first aspect we should check for any new algorithm is correctness. Can you post a summary of how this new approach works? (Either your own description or any work you based it on).
I don't mean to imply that win_eval is 100% correct. It is very naive but captures the simplest expression of the problem. We just need to understand how the new approach makes its decisions.

Second, always share the code you use for performance tests. Especially if the test code is in a different language than that of the final implementation. Matlab optimizes and penalizes different kinds of operations than Python and C++. An algorithm that takes fractions of a second in C++ can easily take over a minute in Matlab if it doesn't match Matlab's preferred design conventions.
In this context, either post the Matlab code used or run the performance tests on a C++ implementation of each.

JNeiger, 2017-02-03:
Instead of representing robots as discrete points, I decided to represent them as a kernal function. This represents the probability that a robot will be at the specific point. I used a [Triweight](https://cloud.githubusercontent.com/assets/5797304/22606954/c0ad41aa-ea23-11e6-9015-3ebb8f131b8b.png) as it produced a smooth curve without an overly complicated formula though I may move to a Tricube to account for the width of the robots. The base width of this function is declared as the sensitivity. It represents how certain the current robot positions will stay constant between now and the time of the kick.

Note that the kernal function that we use has it's Y axis inverted as we want to figure out the probability the pass succeeds.

A set of rays are produced radiating out from the kicking point in the direction of the target. This width can be changed to represent a target segment (in the case of shooting) or just the standard deviation when passing to a specific point. These rays are represented as an delta angle in comparison to the overall kicking target.

Along each angle, we find the distance offset of the robot at the closest point on the current ray. The distance is then scaled by how far away from the kick position the robot is. The result of this function is then thrown into the sensitivity scaled kernal function to get the score. Once all scores have been calculated for this current ray, the minimum is taken. This is mostly due to the fact that we only care about the worst case scenario.

Each ray's minimum is then scaled based off the angle offset from the direct line to the target. These are all summed together and divided by the absolute worst case for all the lines summed together. This result represents the chance it will complete the pass / shot.

[Here](https://gist.github.com/JNeiger/00a2d343e2579aa5c9eaa8071c46da56) is the link to the matlab code. There is some major console spam from the implementation of line intersection. I turned off warnings to stop it, but it still might be happening in the background. I'll try them both in C++ and check out the speed difference there.

jgkamat, 2017-02-04:
I haven't looked at this too much but I think that this algorithm will probably improve our evaluation of passes (#298), but can't replace the window evaluator completely (for the reasons you talked about).

Also is there a paper (or another resource) we can use to look up this approach? If you're thinking that the primary purpose of block chance should be to evaluate better pass probabilities (which we need serious work on), you should assign yourself to #298.

JNeiger, 2017-02-04:
I don't think there is a paper around this. I've been kinda just making it up as I go tbh.

JNeiger, 2017-02-04:
@barulicm 
I just finished converting everything to C++. I only setup the eval_pt_to_pt for each and tested them against the same 6 robot positions for every point on the field. At 500,000 points it is about twice as fast. (865-889 ms for the new algorithm vs 1926-2006 ms for win_eval). At 50,000,000 points it's 87,018 ms and 200,809 ms respectively.

I used 'make tests' to get the time length for each as well as try to normalize any background processes that could affect it.

Linked below are all the files. FYI, I changed the algorithm name to KickEvaluator to try and be a little more descriptive.
KickEvaluator [hpp ](https://github.com/JNeiger/robocup-software/blob/win_eval_replacement/soccer/KickEvaluator.hpp)and [cpp](https://github.com/JNeiger/robocup-software/blob/win_eval_replacement/soccer/KickEvaluator.cpp)
[KickEvaluatorTest.cpp ](https://github.com/JNeiger/robocup-software/blob/win_eval_replacement/soccer/KickEvaluatorTest.cpp)[WindowEvaluatorTest.cpp](https://github.com/JNeiger/robocup-software/blob/win_eval_replacement/soccer/WindowEvaluatorTest.cpp)
[Test1 ](https://github.com/JNeiger/robocup-software/blob/win_eval_replacement/test.txt)[Test2 ](https://github.com/JNeiger/robocup-software/blob/win_eval_replacement/test2.txt)[Test3 ](https://github.com/JNeiger/robocup-software/blob/win_eval_replacement/test3.txt)[Test4](https://github.com/JNeiger/robocup-software/blob/win_eval_replacement/test4.txt)

closed 2017-03-12 by JNeiger