RoboJackets/robocup-software #907: When No Goalie is selected, soccer lags out and previous goalie does things

jgkamat, 2017-02-12:
This cases the processing thread to lag out a lot.

1) We should figure out why it laggs out (this shouldn't happen, ever ideally). I think it's related to two robots trying to move to the same exact spot. 
2) We should prevent the previous goalie from moving around if it's set to None.

The printouts should stay though, they won't lag soccer out that much, and we should have an indication something went wrong.

chachmu, 2017-02-13:
Its possible that this was introduced with the fix that I added to allow plays to continue without a goalie. Someone else should check since I can't tell if the lag comes from my VM being slow or an actual issue

jgkamat, 2017-02-13:
Before your fix it was impossible to run anything with no goalie selected, so I guess it was 'introduced' by that? I don't think it's related to your code though.

EdwardTan20, 2017-02-26:
The issue may not be with goalie assignment but rather with movement. Since i'm running a VM, the difference in frame rate may be skewed

jgkamat, 2017-02-26:
This bug is much more subtle than I thought, and might not be related to the goalie at all (and maybe the goalie only makes it more likeley). I'll need to figure this out myself at some point.

jgkamat, 2017-02-27:
I can't reproduce this on the release version of soccer (with compiler optimization enabled) so I don't think this is too big of an issue. Closing.

jgkamat, 2017-02-27:
never mind, I've just found the root cause (I think). Robot 0 is assigned to two behaviors! :cry: 

[image]

jgkamat, 2017-02-27:
This might reveal a much larger problem in our role assignment system. I think that if a robot is free, and two tactics need a robot at the same time, a robot could be assigned to two tactics at the same time. I'm bumping the priority a bit (since I've never seen this before) but this definetly needs to be investigated.