RoboJackets/robocup-software #912: Add option to get data from StripChart

mmurley3, 2017-02-22:


mmurley3, 2017-02-22:
@ashaw596 I've implemented an "Export Chart" option to the menu bar that can access the data of the current chart.  Can you clarify how/what file type you want the data to be accessed?

ashaw596, 2017-02-22:
Awesome! I think a csv file would be pretty good for this.
Timestamp,pos.x
123123,1.04
123124,10.2
....

jgkamat, 2017-02-22:
I'm a little confused what this issue means. Is this the chart that we can chart changes in our properties with? Will this export all the data from the beginning of time? Also, we should make sure the units are properly setup (I'm not convinced they are right now).

ashaw596, 2017-02-25:
Yeah those charts. Um Maybe? We need to decide that.

mmurley3, 2017-03-01:
Ok, the way I have implemented it right now basically writes the contents of the "history" variable to a .csv file, and while it works fine it seems like this "history" variable only contains the currently viewable part of the graph - not a huge amount.  And of course, the units on all of the data (including time, velocity, and position) are wonky and I'm not sure what they're supposed to be.

chachmu, 2017-03-01:
I think the "history" is somehow linked to our logging system so if someone who knows more about that could take a look at this, that would be great. I think the information is in there somewhere, we just aren't accessing it quite right.

jgkamat, 2017-03-01:
I'm not sure if the log would contain the stripchart history or not, my first guess would be that it dosen't but it's definetly worth to check.

chachmu, 2017-03-01:
You're probably right, having taken a closer look it seems like history is just a vector of log frames that are added "manually" so I don't think it contains any information it isn't given. We might be able to make it not drop previous frames once they pass off the screen though.

chachmu, 2017-03-01:
Also, @mmurley3, I think I found an easy way to find which part of the graph is highlighted if you want to add that functionality. It looks like this [line](https://github.com/RoboJackets/robocup-software/blob/b657690bc06eb6bb5aa99cf618387824f94766bf/soccer/StripChart.cpp#L72-L73) checks which part is highlighted.

PR #916 (2017-06-22): Export Graph Data
Adds option to export current graph's data points to a .csv file.  Currently units are not defined well.
Closes #912
closed 2017-06-22 by chachmu