RoboJackets/robocup-software #921: Switch from Win_Eval to Kick_Eval when calculating chance

JNeiger, 2017-03-12:
We should switch from win_eval to kick_eval when we want to calculate the chance of a shot or pass completing.

jgkamat, 2017-03-14:
We have to be very careful when doing this, since a lot of the defense relies on this. The new method should be better, but it needs to be properly checked to make sure there are no bugs/problems that make performance worse.

JNeiger, 2017-04-13:
What is a good procedure to test it? I can brute force every combination and check for irregularities, but I feel like there is a better way

jgkamat, 2017-04-14:
I think we should be able to manually test all the main cases (And maybe edge cases too). Hopefully we see near-identical results between the old and the new plays.

closed 2019-08-25 by JNeiger