RoboJackets/robocup-software #923: TODO role_assignment.py

gtrosawa, 2017-03-16:
# TODO arbitrary cost lambda property

jgkamat, 2017-03-16:
This is actually a good issue, despite the lack of documentation. 

Essentially, we would want to have plays be able to pass in a lambda expression to define costs, so that they can tweak the cost function however they like! This is a nice issue for people interested in learning more about role reqs.

chachmu, 2017-03-16:
I thought John was already working on that

jgkamat, 2017-03-16:
I'm pretty sure john was working on one of these things:

1. Behavior trees
2. stopped plays schenanigans
3. Lambda expression to prevent restarting of state machine when robot changes.

Although if he is, we should close this as a dupe of the issue he is working on. @jcarn

jcarn, 2017-04-10:
I kind of want this, if @mmurley3 is done with the chart soon I'd like him to work on this. Or anyone really, or I could do it

chachmu, 2017-04-10:
I think the stripchart issue can be finished up this week or next.

jgkamat, 2017-09-12:
moved to https://github.com/RoboJackets/robocup-software/issues/1018#issuecomment-328639620

closed 2017-09-12 by jgkamat