RoboJackets/robocup-software #952: Teleporting ball into a goal in Simulator causes it to disappear

chachmu, 2017-03-16:
If you click inside a goal box to teleport the ball there the ball will dissapear and the robots will no longer be able to interact with it. They continue to act like it is there but are unable to capture it. If you put the ball close to the goal and then teleport it this doesn't happen so I'm not really sure whats going on. If this is on our simulator side then it should be solved by moving to grSim but if its caused by the way we move the ball then that won't help.
[image]

jgkamat, 2017-03-16:
This only happens when the ball is off the field (not in the white rectangle). I think this is a simulator bug and will be fixed by grSim, but I'm not sure. We'll revisit this after that merge.

jgkamat, 2017-07-03:
Hopefully fixed by #904

chachmu, 2017-07-06:
Not fixed by #904.

jgkamat, 2017-07-06:
Welp, I guess that means it's probably not a simulator bug... 😢

JNeiger, 2019-01-18:
Have we seen this recently? I can't say that I've every seen this

chachmu, 2019-01-19:
It doesn't happen super often since you have to pretty intentionally move the ball in a certain way. I'll check tomorrow

chachmu, 2019-02-10:
It still happens but it seems that it will disappear if you teleport it anywhere in the goal not just to the corner