RoboJackets/robocup-software #954: Make Soccer Halt play and stop logging when paused

chachmu, 2017-03-29:
When looking back into the log history using soccer it continues to run plays and log in the background. When the pause button is activated Soccer should Halt play and stop logging to prevent keep the logs from continuing on when they run out of space while someone is trying to look at something. It will probably also be necessary to make it so that restarting play using the normal or force start buttons also restarts logging and pushes the user to the current frame.

jgkamat, 2017-03-29:
Also, I'm debating if this should be an option or not. Probably we should lean on the safe side and say yes unless there's a specific reason not to.

chachmu, 2017-03-29:
I feel like we are getting to the point where we need a config file of some sort so people can have their own defaults. Especially since we have themes now.

jgkamat, 2017-03-29:
We can keep going the environement variable route, which would be easier, but I would love to see a real user-based preference system in place. =D

closed 2020-06-01 by kylestach