RoboJackets/robocup-software #957: Port Robots_Per_Team constant into python, and refactor all magic number 6s

jcarn, 2017-04-04:
Right now we just use "6" in a few places. Magic numbers are bad, and we should replace them. There is a constant called "Robots_Per_Team" in Constants.hpp that would be perfect for this, but it is in the C++ section of the code. Use the robocup-py.ccp stuff to get that constant into python, and then change all 6s to that constant

jcarn, 2017-04-06:
Actually we could use len(main.our_robots()) if we wanted

jgkamat, 2017-04-06:
`len(main.our_robots)` gives us the current number of robots we have (which we usually want) but the c++ constant gives us the max number of robots on a team (which is what a few plays, like stopped, need). I think we could make a function for `len(our_robots)` but we can't make it a constant since the number of robots we have might change in a match.

longoodr, 2017-11-19:
@mmurley3 Are you still working on this?

mmurley3, 2017-11-19:
Nah, I haven't had the time to make much progress on it. Feel free to tackle it if you want.

Jason27chan, 2018-07-17:
@jcarn Which files should we be looking at?

jcarn, 2018-07-18:
This was more of an intro issue I had in mind

kylestach, 2019-08-29:
@tjones320 this is a must-have before competition this year.