RoboJackets/robocup-software #964: Normalizing Angles

Jason27chan, 2017-04-12:
Some places (circle_near_ball being on example) use a weird way to normalize angles when it should use Geometry2D's

Jason27chan, 2018-07-17:
@jgkamat I cant find a function to normalize angles in Geometry2D. Can we close this issue or make this an issue to make a function to normalize in Geometry2D?

JNeiger, 2018-07-17:
https://github.com/RoboJackets/robocup-software/blob/master/common/Utils.hpp#L57
Is this maybe what we were talking about?

ashaw596, 2018-07-17:
Also since its a common use case. https://github.com/RoboJackets/robocup-software/blob/8871c0ea6c982ccbc527ae5695441ed1ff0f8d8a/common/Geometry2d/Point.hpp#L345
This should be used to get a normalized angle between directional angles.