RoboJackets/robocup-software #967: Add one playbook to gitignore

jcarn, 2017-04-12:
A lot of testing involves running one play repeatedly, and often requires restarting every time. It is quicker to set up a playbook with that play selected and designate as the playbook to use. However, it is annoying to deal with the git part of changing a file you don't want to commit. Could we create a playbook like testing_playbook that we add the the ignore?

jgkamat, 2017-04-12:
Maybe we could just have soccer start with a personal_playbook.pbk file by default, and if it dosen't exist, nothing get's loaded. It would be hard to advertise and document though...

We could then add that playbook file to the gitignore, so people can add it if they want (and it won't get committed).

jcarn, 2017-04-12:
This could be a good issue for Fall next year, unless I get really bored I'll probably let it sit until then

jgkamat, 2017-07-22:
Ok I'm wasting a bunch of time on this so this is happening now.

commit e437d531 (2017-07-22): Add a default playbook file

Closes #967
PR #996 (2017-07-22): Add a default playbook file
Closes #967

This is a little different from what we discussed previously, it's a file that's added (that I'm trusting people not to commit) instead of just an addition to gitignore. I think this will be more helpful because it includes usage instructions.

Let me know if you think it should be implmeented differently, I'm going to merge this soon since it's a tinychange and I want it in :smile:
closed 2017-07-22 by jgkamat