RoboJackets/robocup-software #969: Add a max_robot_change_cost constant

jcarn, 2017-04-15:
It would be handy to have a constant that would allow us to pass it as the robot_change_cost param for role_reqs. The constant would basically say "don't even try switching robots for this behavior. This would be useful e.g. if we wanted a complex movement around the field, like to navigate an offensive robot around defenders. 

If this didn't make sense I'll spend more time rewriting the issue so its clearer

jgkamat, 2017-04-15:
Maybe this can be a lambda expression instead? #functional ;)

jcarn, 2017-04-19:
Yeah it should probably be a lambda expression instead of just a number. 

Also there is a required_robot_id field in role_reqs so the functionality of a max change constant would probably be better implemented through that

acpollmann, 2018-01-10:
This can already be achieved by setting max_robot_change_cost constant to infinity.

closed 2018-01-10 by acpollmann