RoboJackets/robocup-software #971: Arch-Setup: "python setup.py egg_info failed with error code 1"

JNeiger, 2017-04-19:
I was running util/arch-setup and while it is installing stylize, it fails with the following error. Python version is 3.6.0. I'll keep playing around with it to see if it's just my machine or something with the arch-setup.

> -- Installing udev rules
-- Installing required packages
 there is nothing to do
-- Installing required AUR packages
Collecting graphviz>=0.4.2 (from -r /home/joe/robocup-software/util/requirements3.txt (line 3))
  Using cached graphviz-0.6-py2.py3-none-any.whl
Collecting watchdog (from -r /home/joe/robocup-software/util/requirements3.txt (line 4))
  Using cached watchdog-0.8.3.tar.gz
Collecting munkres (from -r /home/joe/robocup-software/util/requirements3.txt (line 5))
  Using cached munkres-1.0.8.tar.gz
Collecting pylint (from -r /home/joe/robocup-software/util/requirements3.txt (line 6))
  Using cached pylint-1.7.1-py2.py3-none-any.whl
Collecting stylize>=0.3.0 (from -r /home/joe/robocup-software/util/requirements3.txt (line 8))
  Using cached stylize-0.3.1.tar.gz
    Complete output from command python setup.py egg_info:
    Traceback (most recent call last):
      File "<string>", line 1, in <module>
      File "/tmp/pip-build-1h6yonmk/stylize/setup.py", line 9, in <module>
        long_description=open('readme.rst', 'r').read(),
      File "/usr/lib/python3.6/encodings/ascii.py", line 26, in decode
        return codecs.ascii_decode(input, self.errors)[0]
    UnicodeDecodeError: 'ascii' codec can't decode byte 0xe2 in position 203: ordinal not in range(128)
    Command "python setup.py egg_info" failed with error code 1 in /tmp/pip-build-v_v59njy/stylize

jgkamat, 2017-04-19:
Can you try running: `pip install stylize==0.3.0`?

It seems like something wrong with unicode, stylize needs unicode. Can you try regenerating your locales: https://wiki.archlinux.org/index.php/Locale

JNeiger, 2017-04-19:
Yep, forgot to generate my locales. Thanks!

closed 2017-04-19 by JNeiger