RoboJackets/robocup-software #973: Split 2D overlay from Adaptive Formation into it own function

JNeiger, 2017-04-25:
See [here](https://github.com/RoboJackets/robocup-software/pull/958#discussion_r112817441)

JNeiger, 2017-05-26:
Where would be a good place to put this? I can't think of a folder anywhere where this would fit in nicely.

jgkamat, 2017-05-26:
I would agree with you that there's no good place to put this right now, how about we make a new module for it? We could call it visualization, or misc?

If we wanted to avoid making another module for some reason though, the best place (in my opinion) would be in ./soccer/gameplay/evaluation.

JNeiger, 2017-08-23:
This could also be implemented into the play class itself. It might be nice to see in the simulator when we run it on real robots or something.

PR #1008 (2017-09-24): Added 2d overlay functions to simplify drawing
Closes #973. I decided to ditch the play class function way. This is a lot simpler to implement and deal with I think
closed 2017-09-24 by JNeiger