RoboJackets/robocup-software #978: Write docstrings for python

jcarn, 2017-04-29:
A lot of plugins python docstrings to provide information about imported modules, but right now we don't have any afaik. We should add these, it would be a great way to learn the codebase!

jgkamat, 2017-04-30:
This is actually an interesting point that we might want to consider in more detail. Right now, our docs are generated from `##` at the top of methods, but we should really be taking advantage of the official python documentation so we can view it in the interpreter as well as online. We need to see if doxygen supports this style of documentation though, and if it does, we probably should convert everything to use the official docstring style.

kylestach, 2019-08-22:
It looks like doxygen does support docstrings, although it does not support special doxygen commands `\something` inside of them. It doesn't look like we actually use those anywhere though, so we should be good to switch.

closed 2022-07-12 by kasohrab