RoboJackets/robocup-software #985: Verify/Implement ball placement both when placing and when standing by

jgkamat, 2017-06-08:
Our current stopped play will conflict with this rule:

[image]

We should probably:

1) go into line up when a ball placment is issued to the opposing team
2) Implement #676 

@chachmu do you think you could take this?

chachmu, 2017-06-19:
Currently our robots just freeze in place if the other team is supposed to place the ball. I can fix that and I think I will also need to change how exiting the ball placement state works since they no longer specify how play is restarted after the placement.

chachmu, 2017-06-19:
@jgkamat Do you know where the SSL coordinate system for the field is defined? I want to find an official source so I know I get it right this time but it is proving surprisingly difficult to find.

jgkamat, 2017-06-19:
The docs seem surprising hard to figure out... [here it says they use 'ssl-vision' coordinates](https://github.com/RoboCup-SSL/ssl-refbox/blob/ead58770b032ca380987cb9e1200b04b0d1db0f3/rcon.proto#L56-L62)

I'm not sure what that means, but I think that means that (0,0) is the 'bottom of the field'. One teams goal is the 'bottom' in SSL Vision (and we flip it using +/- defense internally). We probably want to get the raw, unflipped coordinates from ssl-vision.

chachmu, 2017-06-21:
Yeah I need to find what format the SSL vision system uses because I am pretty sure last year the x and y coordinates were flipped too. I think it was something like we say that x is the width of the field and y is the length but SSL has it opposite of that.

PR #987 (2017-07-23): Update ball placement to work with new rules
- [x] Move out of the way of ball placement for other team
  - [x]  Move out of the way of own ball placement?
- [ ] Fix coordinate system to match SSL standard
- [ ] Increase speed of placement
- [x] Fix issue where robot goes to random point outside of circle for a second after placing, sometimes running into the ball

Closes #985 and Closes #676.
PR #1001 (2017-08-25): Merging Changes from Competition 2017
Let's get this started.

Common: https://github.com/RoboJackets/robocup-common/pull/15
Firmware: https://github.com/RoboJackets/robocup-firmware/pull/40

Blocking #911 

Closes #985, closes #676
closed 2017-08-25 by joshhting