RoboJackets/robocup-software #986: Problem with RF Communication

maxaubel, 2017-06-10:
Hey guys, 
We get the following error when trying to connect thru an RF Antenna using the soccer software. See the captures for more details regarding to the errors we have.
Hope you can help us
Bye!

[image]
[image]

jgkamat, 2017-06-10:
Hi @maxaubel, thanks for trying out our software! We hope you like it :). 

I think that we might need a little more information to try and debug this. The `LIBUSB_ERROR_IO` sounds like a problem writing/finding the radio, so you might want to start by installing the [udev rules that we use](https://github.com/RoboJackets/robocup-software/blob/master/util/robocup.rules) into your machine. I would also try running soccer as root (just to try and see if this is a permissions issue or not, if that works, let us know and we can try to find a solution for that).

If that does not work, could you tell us more about the radio hardware you are using? We haven't tested any of our setup with normal radio base stations, as we use our own custom boards, so it is very likeley our software does not support the radio base station you are using (and we, or you, would have to add support for it). 

Let us know what you find out, and we'll try to help you! :smile: 

Also pinging @petersonev, @guyfleeman, do you guys have any ideas?

Let us know if you run into any other issues while you're using soccer, and we'll try our best to solve them! :tada:

maxaubel, 2017-06-10:
Hi! Thanks a lot for the quick response!! 
We aren't actually using any special hardware, we're just using a connected pair of FTDIs. The issue could be just that we're not sending any bit of data. We're just testing if the connection works. The program's radio indicator 'blinks' (it shows Radio Disconnected and Radio Connected). 

[image]

And over it, it shows 'NO RADIO RX' over it.

Thanks a bunch guys, bye!

jgkamat, 2017-06-11:
Just to start off, most of the code that interfaces with the radio is in USBRadio.cpp (https://github.com/RoboJackets/robocup-software/blob/b1cca74b62e0c4135c8f17bb51bba24aad5ad84a/soccer/radio/USBRadio.cpp), so it might make sense to try looking in there to solve this problem.

I'm not exactly which radio you're referring to when you say FTDI, but we use our own custom radio hardware, and we haven't tested our software with any other radio system (so I'm not sure how easy or hard it would be to get this working). On top of that, the protocol we use for controlling the robots is entirely our own as well, so unless you were able to read that protocol in your own firmware, getting this set up wouldn't have much benefit. **Could you give me the *exact* model number of your radio?** That might help me get started in looking at this, (and depending on the type, I might be able to buy one, which would make it much easier for me to look at this).

We use [libusb](https://github.com/libusb/libusb) to access our radios, and the code where we initialize radios is [located here](https://github.com/RoboJackets/robocup-software/blob/b1cca74b62e0c4135c8f17bb51bba24aad5ad84a/soccer/radio/USBRadio.cpp#L53). [On this line](https://github.com/RoboJackets/robocup-software/blob/b1cca74b62e0c4135c8f17bb51bba24aad5ad84a/soccer/radio/USBRadio.cpp#L56) (line 56), we filter out any radios we haven't tested, so I would recommend either removing that check or finding the vendor/product ID of your radio, and replacing that check to verify your vendor/product ID. 

If that dosen't do the trick, Can you check to see if any of the following messages [printed from this code here](https://github.com/RoboJackets/robocup-software/blob/b1cca74b62e0c4135c8f17bb51bba24aad5ad84a/soccer/radio/USBRadio.cpp#L68-L98) show up as soon as soccer starts up (error messages that are printed when the device initialized):

* USBRadio: No radio is connected
* USBRadio: All radios are in use
* USBRadio: Can't set configuration
* USBRadio: Can't claim interface

Lastly, the error message that you are actually seeing is printed [from this bit of code](https://github.com/RoboJackets/robocup-software/blob/b1cca74b62e0c4135c8f17bb51bba24aad5ad84a/soccer/radio/USBRadio.cpp#L236). Unfortunately, if anything wrong happened in setting up the radio whatsoever, you will see that one generic error message. Because I don't have your radio hardware with me, it makes it very hard for me to remotely debug this (since I can't try anything out for myself), so I'm sorry that I'm unable to do more to try to fix this. Let me know if there's anything that's confusing and I'll try to explain it in more detail.

maxaubel, 2017-06-17:
Hey guys!
We kept trying to solve this issue, with no solutions. We use an usb to serial conversor named FTDI FT232RL. Also, the model of the radio we use is APC220. We get errors with libusb_set_configuration and libusb_claim_interface (we tried commenting both if's, but an error still ocurred).

Thanks for all your time guys!

maxaubel, 2017-06-17:
Actually we fixed the error related to libusb_set_configuration and libusb_claim_interface downloading a newer version of libusb. But we are still getting the error message when trying to send a forward packet and it prints "UBSRadio: Bulk write failed sent = 0, size = 62".

jgkamat, 2017-06-18:
@maxaubel after fixing the set_configuration and claim_interface (which seems like a good step forward), do you get any printouts from [this section of code here](https://github.com/RoboJackets/robocup-software/blob/d91e6f05c83de5ef771174b1b58117a4881aab0f/soccer/radio/USBRadio.cpp#L238-L240) (is the `transferRetCode` variable == `LIBUSB_SUCCESS`)? If the return code is not `LIBUSB_SUCCESS` that print statement should give us more information about what could be wrong (failed permissions, failed write, etc), which could help us figure out what libusb thinks is wrong at this point. I would also try (just for a test) to see if running `soccer` as root fixes this (which would tell us that the problem was specifically a permissions error). 

I did a little googling to see if anyone has used the usb-serial converter you mentioned with `libusb` and I couldn't come up with any results. If you could write a small example of writing to your FTDI converter and radio with `libusb` I could take a look at it and see if we do anything different (which would help immensely!). 

I'm not sure about this, but you might need [this library](https://www.intra2net.com/en/developer/libftdi/) to actually communicate with the FTDI board (and you would need to write code specifically to use it), but I'm not sure if that's required or not. 

Let me know what you find out, and I'll try to give you more information! :smile:

fpintoG, 2017-07-15:
Hello
After all, we realized that the RF communication is actually working, nevertheless, it does not build the packet cause it requires the id of every single robot given by the ssl-vision program (I'm not sure about that) and now we can detect just the team position but not the ids.

jgkamat, 2017-07-15:
That's great to hear! If you can get some data working, it will probably be a lot easier to debug the rest!

Our communication does depend partially on SSL Vision data being available (I should have mentioned that before, sorry!). SSL Vision should properly be recognized by our program, and it should get the ID's and such properly. If the problem you're having is an bad connectiont to your SSL Vision, let me know and I'll take a look at that.

If you want to do some testing without SSL Vision, you can turn on our 'manual' mode, which let's you drive robots without any SSL Vision at all. Here's a few steps to try to help you along with that:

1. Select an ID from the 'Manual drop down'. Please be careful with any robots you might be using at this point (since if there are any bugs in soccer, it could cause your robots to run out of control). Please make sure you select any ID except `None`
[image]

2. Plug a joystick into the computer running soccer. Once the joystick is recognized, a message will be printed to the console. To verify the joystick is working, you can go to the joystick tab on the right (pictured), and move the stick around, and see if the input is recognized:
[image]

3. At this point, you should be able to move robots around by moving the joystick. Since our robot protocol is probably different from yours, this might not work though. In any case, attempting to move the robots should force our program to send some packets.

Hopefully these help you in your debugging, sorry about not being able to help more... 

As always, let me know if you have any additional questions or if you get stuck anywhere else! :smile:

jgkamat, 2017-07-27:
@fpintoG, @maxaubel 

I thought I would add that I'm currently at RoboCup 2017 in Japan, and I would love to talk to you and help you out in person, if you're here too! If you're at RoboCup 2017, I'm with RoboJackets in the Small Size League.

Finally, let us know if you have any other questions, we love to help new teams get into Small Size League, since it's quite a leap for beginners, and it's really exciting to see another team benefit from our work! :smile:

maxaubel, 2017-07-28:
Thanks for the invite, but we couldn't make it to this year's RoboCup :(. But we are working hard for February's competition. Oh and by the way, the ssl-vision git guys aren't as active as RoboJackets, so maybe you could have a solution for another issue we have in that regard. https://github.com/RoboCup-SSL/ssl-vision/issues/76

Thanks a lot for all your time and have success in Japan!!

jgkamat, 2017-09-03:
I'm going to close this issue to clean up our issues list. If you have any more questions about `robocup-software`, feel free to reopen this issue, open a new issue here, or simply tag me (@jgkamat) anywhere on github. Thanks again for getting in touch with us! :smile:

closed 2017-09-03 by jgkamat