RoboJackets/robocup-software #988: Refactor worldToTeam and other transformations

chachmu, 2017-06-21:
Currently the worldToTeam transformation matrix is calculated in at least 2 different places and uses a slightly different method each time. This needs to be refactored so that it is only defined in one place. Currently Processor, FieldView, and GameState use the matrix.

chachmu, 2017-06-21:
I'm not really sure where the best place to move everything to that would make the most sense. @jgkamat any suggestions? I could make a new class to store it but that seems a little overkill.

jgkamat, 2017-06-22:
@chachmu I think that the best place to have it is probably FieldView (that makes the most sense to me). I haven't looked at it too  much, but if GameState seems easier, that also seems fine. Please keep it out of processor though 😄