RoboJackets/robocup-software #1000: 1k Issue Party

jgkamat, 2017-07-28:
TODO:

Plan a 1k Issue party
Execute a 1k Issue Party
Close the 1k'th issue

joshhting, 2017-08-22:
We should discuss this today

joshhting, 2017-08-23:
Option 1: Rocket league

joshhting, 2017-08-23:
Option 2: Factorio

joshhting, 2017-08-23:
Option 3: Mooch off of RJ social

joshhting, 2017-08-23:
Option 4: Card games

joshhting, 2017-08-23:
Option 5: Crash an IGVC meeting

jgkamat, 2017-09-06:
Looks like we'll be playing factorio.

Right now, my best idea is to schedule this for a saturday, probably before software training (6PM). Anyone have any times that they prefer more?

jgkamat, 2017-10-16:
This event will take place from 1-4PM in IC205. :tada:

ashaw596, 2017-10-16:
lol. +D Today?

jgkamat, 2017-10-16:
Whoops, I'm so bad at this, I updated the comment.

joshhting, 2017-10-19:
Are we booking the room for this?

jgkamat, 2017-10-19:
Yup, the room should be booked (IC205)

joshhting, 2017-10-22:
@jgkamat So Jason & John can't make that time. Could you send out a whenisgood to determine the optimal weekend for this?

jgkamat, 2017-10-22:
We've already booked the room, and I don't think it's a good idea to cancel/re-book the room .

From earlier:
>Right now, my best idea is to schedule this for a saturday, probably before software training (6PM). Anyone have any times that they prefer more?

I'm hoping that this can lead into an informal factorio server we can play on whenever.

FWIW, it's possible to 'attend' remotely, since we'll be just playing factorio.

I think it's a better idea to just do another one next semeser or another one in a couple weeks, re-scheduling will just add confusion IMO.

jgkamat, 2017-10-25:
Because of scheduling mishaps, the location has been moved to **CoC 102** instead of the IC. I'm updating the previous post as well, so people don't get confused. Perhaps we should send an email to the robocup mailing list too :P.

joshhting, 2017-10-28:
@jgkamat would you like to do the honors

closed 2017-10-29 by jgkamat