RoboJackets/robocup-software #999: Rewrite Defense.py

jgkamat, 2017-07-28:
While the logic in Defense.py is mostly good, the file is really unmaintainable, not because it has no documentation, but because it's very poorly organized (and is kind of a mess).

When we rewrite it, we should consider using adaptive formation to improve defense placement.

JNeiger, 2017-07-28:
Are you talking about using the defensive risk stuff that's already implemented or doing another probability style play specifically for defense?

jgkamat, 2017-07-29:
I wanted to primarily rewrite defense to make it more readable.  If we determine the provability stuff is better we should switch to that in the future, our existing defense plays are no longer effective.

Er force's TDP has more info

JNeiger, 2017-08-23:
Here's a list of the different weights we and Er force use.

Er force
- Distance to our goal
- Ball-opponent-goal angle (One touch shot angle)
- Ball-goal-opponent (Get defender travel dist)

Us
- Pass to opponent
- Time until activation (Time until opponent will have the ball)
- Position (Weighted by distance to goal, angle to goal, and distance to center line)
- Space near the point
- Ball-opponent-goal angle (One touch shot angle)
- Distance to ball

JNeiger, 2017-08-28:
Does the algorithm that chooses the optimal positions for the goalie and defenders to block a incoming shot need to be improved too?

jgkamat, 2017-08-28:
I think that blocking an incoming shot is handled well with the existing setup, our problem there is in motion control. We catch nearly every shot in the simulator, as far as I've tested.

joshhting, 2017-09-03:
Refer to #256 for related discussion as well

JNeiger, 2017-10-24:
Going to keep this open while I see if I can improve what we have now

JNeiger, 2019-01-18:
Closing this since the original task has been finished.

PR #1011 (2017-10-24): Defense rewrite
Closes #999
closed 2019-01-18 by JNeiger