RoboJackets/robocup-software #1014: Clear 'unselected' plays during a playbook load

jgkamat, 2017-09-03:
When loading a playbook, new plays are added, but checked plays are not deleted. This caused a small problem at comp, and I really don't want to have to deal with that again.

Bonus: Add a indicator on the plays panel showing if the playbook was modified manually (in the program) or if it's freshly loaded from the file.

krispreza, 2017-09-06:
In your above comment you say that checked plays are not deleted. I am assuming that you don't actually want to delete the plays from the play registry but just want to uncheck/disable them right?

jgkamat, 2017-09-06:
yup, I just want them restored to 'unchecked'. Deleting them would probably be overkill.

Sorry for the confusion :)

PR #1023 (2017-10-15): now restores checked plays to unchecked when loading new playbook
Closes #1014
closed 2017-10-15 by JNeiger