RoboJackets/robocup-software #1016: LIBUSB ERR Timeout

jgkamat, 2017-09-03:
Arguably our biggest issue during comp2017.

```
USBRadio: Bulk write failed. sent = 0, size = 62
  Error: 'LIBUSB_ERROR_TIMEOUT'
```

guyfleeman, 2018-06-03:
Okay are there any updates on this? @jgkamat

jgkamat, 2018-06-03:
I haven't been able to reproduce this while testing in the past year, but im nowhere sure if this is solved or not. We need to test with 6 robots to try to reproduce it...

guyfleeman, 2018-06-04:
@ashaw596 I still want to look at it since I think we intend to start sending back more info from motion control related things. Was debug enabled at comp?

ashaw596, 2018-06-04:
We disabled it during matches since we were having so much radio
instability with 6 robots already. The base station has some weird serial
USB driver problems probably. Not 100% sure but wasn't someone working on
an actual debug communication protocol(that wasn't so Jacky). I never
merged the debugCommunication branch.

JNeiger, 2019-07-20:
Killing this since we swapped to network radios

closed 2019-07-20 by JNeiger