RoboJackets/robocup-software #1021: Switch translation and rotation joysticks for manual mode

joshhting, 2017-09-20:
Pretty much anyone who has used a controller for any shooter, action or racing game has developed the intuition of translation for the left joystick and rotation for the right joystick.

jgkamat, 2017-09-20:
This is a bit harder than the typical intro issue, but I think should be fine for anyone ready to do a tiny bit of extra work! 

This can be fixed by tweaking values on [these lines](https://github.com/RoboJackets/robocup-software/blob/6b05a2436f8005956c5823b72d2a2aa671ac3aaf/soccer/joystick/GamepadController.cpp#L164-L180) and [these lines](https://github.com/RoboJackets/robocup-software/blob/d7768ec2647940f70fa849ad0f22d699033abb52/soccer/joystick/GamepadJoystick.cpp#L94-L100). Please test them with real joysticks though! :).

chachmu, 2018-03-13:
This was merged a while ago, this issue just wasn't linked to the pull request.

closed 2018-03-13 by chachmu