RoboJackets/robocup-software #1027: grsim Headless Mode

chachmu, 2017-09-26:
A lot of people are having issues with performance on VM's due to grsim. We actually need a headless mode now.

jgkamat, 2017-09-26:
This will require a patch to grsim, unfortunately. I haven't looked into it, but I really we should try to do it.

ashaw596, 2017-10-08:
So I've been messing a bit with performance. I seem to gett 60 fps if you get set isGLEnabled = false; on line 142 of external/grSim/src/sslworld.cpp to false. This disables the opengl display completely. and set change line 171 in external/grSim/src/pworld.cpp to dWorldQuickStep from dWorldStep.

The first change is just cosmetic and performance boost since we get rid of most of the visualization overhead. The second one actually changes the quality of the physics simulation. We'll need to test to see how it affects the simulation quality.

ashaw596, 2017-10-08:
The header for those 2 functsion is this btw.
/**
 * @brief Step the world.
 *
 * This uses a "big matrix" method that takes time on the order of m^3
 * and memory on the order of m^2, where m is the total number of constraint
 * rows. For large systems this will use a lot of memory and can be very slow,
 * but this is currently the most accurate method.
 *
 * Failure result status means that the memory allocation has failed for operation.
 * In such a case all the objects remain in unchanged state and simulation can be
 * retried as soon as more memory is available.
 *
 * @param w The world to be stepped
 * @param stepsize The number of seconds that the simulation has to advance.
 * @returns 1 for success and 0 for failure
 *
 * @ingroup world
 */
ODE_API int dWorldStep (dWorldID w, dReal stepsize);

/**
 * @brief Quick-step the world.
 *
 * This uses an iterative method that takes time on the order of m*N
 * and memory on the order of m, where m is the total number of constraint
 * rows N is the number of iterations.
 * For large systems this is a lot faster than dWorldStep(),
 * but it is less accurate.
 *
 * QuickStep is great for stacks of objects especially when the
 * auto-disable feature is used as well.
 * However, it has poor accuracy for near-singular systems.
 * Near-singular systems can occur when using high-friction contacts, motors,
 * or certain articulated structures. For example, a robot with multiple legs
 * sitting on the ground may be near-singular.
 *
 * There are ways to help overcome QuickStep's inaccuracy problems:
 *
 * \li Increase CFM.
 * \li Reduce the number of contacts in your system (e.g. use the minimum
 *     number of contacts for the feet of a robot or creature).
 * \li Don't use excessive friction in the contacts.
 * \li Use contact slip if appropriate
 * \li Avoid kinematic loops (however, kinematic loops are inevitable in
 *     legged creatures).
 * \li Don't use excessive motor strength.
 * \liUse force-based motors instead of velocity-based motors.
 *
 * Increasing the number of QuickStep iterations may help a little bit, but
 * it is not going to help much if your system is really near singular.
 *
 * Failure result status means that the memory allocation has failed for operation.
 * In such a case all the objects remain in unchanged state and simulation can be
 * retried as soon as more memory is available.
 *
 * @param w The world to be stepped
 * @param stepsize The number of seconds that the simulation has to advance.
 * @returns 1 for success and 0 for failure
 *
 * @ingroup world
 */
ODE_API int dWorldQuickStep (dWorldID w, dReal stepsize);

chachmu, 2018-03-11:
Waiting for us to add support for the square field so we can switch to the new branch of grsim and change the flag to the correct one.

PR #1114 (2018-04-08): Add a make option to run grsim with a window
If you don't want to run grsim in headless mode then you can call `make run-headmore`. I blame Jay for this atrocity. Closes #1027.
closed 2018-04-08 by chachmu