RoboJackets/robocup-software #1034: Flash Entire UI Red when the gameplaesy system crash

ashaw596, 2017-10-03:
Right now, if there is a bug, the system just restarts and outputs in the command line. We should make a bigger indicator such as this.

jgkamat, 2017-10-17:
We could use libnotify to send an urgent notification to the system notification daemon, which would be more 'user friendly' I think.

CollinAvidano, 2017-11-29:
I agree with @jgkamat. If its fine with you all I will implement the system notification.

jgkamat, 2017-11-29:
Also, these probably need to have an option to turn this off for debugging. While testing, it's common to have exception loops, and we don't want to attack people's notifications. At least we should throttle or digest these.

Also see #591, which is basically a duplicate of this.

jgkamat, 2017-11-29:
Also, the command for interacting with system notifications is notify-send.

Ex: `notify-send "hello" "world"`

however, if you find a 'libnotify' interface to python, that might be easier to use.

closed 2022-07-12 by kasohrab