RoboJackets/robocup-software #1089: Angular velocity

maxaubel, 2018-01-12:
This is not an issue, rather a question. In USBRadio.cpp, when you guys get the angular velocity for a certain robot (https://github.com/RoboJackets/robocup-software/blob/4d9222e476a52017e8c1a605b85a208851453374/soccer/radio/USBRadio.cpp#L223). What's the unit for "robot.avelocity()"? If I print it, I'm getting values like 0,008727, which is too small for being degrees/sec or rad/s. I tried searching for the origin of this value, but I couldn't find any unit related stuff. Thanks for your time.

JNeiger, 2018-01-13:
I believe it is `1/1000 rad/s`. Multiplying it by `VELOCITY_SCALE_FACTOR` should be it back to just rad/s.

maxaubel, 2018-01-13:
Yeah the simulator is sending the scaled velocity but it gets divided by 1000 again in the firmware meaning that the robot reach angular velocities around 4 grades per second.

JNeiger, 2018-01-14:
So I just took another look at this and everything I've checked out, including `simradio`, uses `rads/s`. I unfortunately do not have an `USBRadio` to test this specifically, but since everything else is `rads/s`, I think it is safe to assume this is too.

closed 2018-01-31 by chachmu