RoboJackets/robocup-software #1091: Ball speeds IRL differ from speeds seen in soccer

JNeiger, 2018-01-24:
I think you can simulate this by changing the kick_strength to a constant and measuring the velocity using the tree tab.

JNeiger, 2018-01-24:
`vel_vision_bug` is the branch I made that produces these problems. We set the velocity of the kick to 5 in the simradio so any velocity we read should be at 5 m/s. The TestKicker play should just print out the ball velocity while trying to shoot on the goal.

chachmu, 2018-01-28:
I tested the kick speed from python using `main.ball().pos` and it gave the correct result so I'm pretty sure the positions reported are correct meaning that the issue is somewhere in the actual velocity calculation in c++ which is in [ballFilter.cpp](https://github.com/RoboJackets/robocup-software/blob/6731e7a74d22cfc5b2c6a4c60c4bbdd094e6fc83/soccer/modeling/BallFilter.cpp#L20).

ashaw596, 2018-01-30:
Yeah theres still a couple different Ball predictions methods. They should be merged at some point. Did you check the speed in the Tree? You can graph the vels.

JNeiger, 2018-01-30:
I'm 90% sure I checked that and it was giving the wrong values too

chachmu, 2018-01-30:
As far as I can tell, it is wrong in every single place that the c++ side lists it.

chachmu, 2018-01-31:
It looks like this issue is caused by the smoothing filter on the ball velocity. By the time the filter starts to reach the velocity of the ball it has decelerated enough that the velocity is much lower than the initial kick speed. This can be fixed either by setting the alpha to 1.0 to remove the smoothing or by completely rewriting the ball filtering and prediction system.

ashaw596, 2018-06-04:
> @guyfleeman <https://github.com/guyfleeman> It depends on what we want to
> do. If we are fine with the smoothing filter making the max speed look
> lower then we can just leave it as is. If we want to make it report the
> actual speed without smoothing the input then the easiest way would be to
> set the alpha value to 1.0

A better fix would be a smarter filter like a Kalman or particle, but
that's complex. A hack to detect kicks/large changes in velocity combined
with alpha tuning might handle the cases we care about better. Or just the
alpha might be enough to work.

JNeiger, 2019-02-03:
Fixed by #1227

closed 2019-02-03 by JNeiger