RoboJackets/robocup-software #1105: Estimated Time Runtime Error

jgkamat, 2018-02-25:
```
Estimated Time can't be before observation time.
terminate called after throwing an instance of 'std::runtime_error'
  what():  Estimated Time can't be before observation time.

Thread 11 "QThread" received signal SIGABRT, Aborted.
[Switching to Thread 0x7fffd37fe700 (LWP 20962)]
__GI_raise (sig=sig@entry=6) at ../sysdeps/unix/sysv/linux/raise.c:51
51	../sysdeps/unix/sysv/linux/raise.c: No such file or directory.
(gdb) bt
#0  __GI_raise (sig=sig@entry=6) at ../sysdeps/unix/sysv/linux/raise.c:51
#1  0x00007ffff2c3c3fa in __GI_abort () at abort.c:89
#2  0x00007ffff35530ad in __gnu_cxx::__verbose_terminate_handler() () from /usr/lib/x86_64-linux-gnu/libstdc++.so.6
#3  0x00007ffff3551066 in ?? () from /usr/lib/x86_64-linux-gnu/libstdc++.so.6
#4  0x00007ffff35510b1 in std::terminate() () from /usr/lib/x86_64-linux-gnu/libstdc++.so.6
#5  0x00007ffff35512c9 in __cxa_throw () from /usr/lib/x86_64-linux-gnu/libstdc++.so.6
#6  0x00007ffff7a2491c in debugThrow<std::runtime_error, 0> (e=...) at ../common/Utils.hpp:37
#7  debugThrow (string="Estimated Time can't be before observation time.") at ../common/Utils.hpp:42
#8  0x00007ffff7aa3d38 in Ball::predict (this=<optimized out>, estimateTime=...) at /home/jay/Code/robocup-software/soccer/SystemState.cpp:66
#9  0x00007ffff7a2136b in BallFilter::predict (this=<optimized out>, time=..., velocityUncertainty=velocityUncertainty@entry=0x0) at /home/jay/Code/robocup-software/soccer/modeling/BallFilter.cpp:27
#10 0x00007ffff7a219b4 in BallTracker::run (this=0x555555814ac0, obs=std::vector of length 1, capacity 1 = {...}, state=state@entry=0x5555557cb2f8)
    at /home/jay/Code/robocup-software/soccer/modeling/BallTracker.cpp:140
#11 0x00007ffff7a7afac in Processor::runModels (this=this@entry=0x5555557cb260, detectionFrames=std::vector of length 3, capacity 4 = {...}) at /home/jay/Code/robocup-software/soccer/Processor.cpp:265
#12 0x00007ffff7a7ec76 in Processor::run (this=0x5555557cb260) at /home/jay/Code/robocup-software/soccer/Processor.cpp:430
#13 0x00007ffff6680da8 in ?? () from /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#14 0x00007ffff5f4b494 in start_thread (arg=0x7fffd37fe700) at pthread_create.c:333
#15 0x00007ffff2cf0aff in clone () at ../sysdeps/unix/sysv/linux/x86_64/clone.S:97
(gdb) 
```

jgkamat, 2018-02-25:
Reproduce by moving the ball around while running defense (only happens rarely for me).

JNeiger, 2018-02-25:
Here are the lines referenced in the stack trace.
[debugThrow<std::runtime_error, 0> (e=...)](https://github.com/RoboJackets/robocup-common/blob/a107d2e7041b9aa493b83acd8b356779426a99b1/Utils.hpp#L37)
[Ball::predict (this=<optimized out>, estimateTime=...)](https://github.com/RoboJackets/robocup-software/blob/master/soccer/SystemState.cpp#L66)
[BallFilter::predict (this=<optimized out>, time=..., velocityUncertainty=velocityUncertainty@entry=0x0)](https://github.com/RoboJackets/robocup-software/blob/master/soccer/modeling/BallFilter.cpp#L27)
[BallTracker::run (this=0x555555814ac0, obs=std::vector of length 1, capacity 1 = {...}, state=state@entry=0x5555557cb2f8)](https://github.com/RoboJackets/robocup-software/blob/master/soccer/modeling/BallTracker.cpp#L140)

chachmu, 2018-03-11:
I know we are considering rewriting the ball filter, if we did then this issue wouldn't really matter anymore

chachmu, 2018-03-11:
I'm making a workaround that logs a critical error instead of crashing for now so if we don't have to worry about forgetting about this and crashing at comp

chachmu, 2018-03-11:
Turns out there isn't an error logging system for C++ right now so it just prints out a very stern warning

Jason27chan, 2018-03-13:
CRITICAL ERROR: Estimated Time can't be before observation time
estimateTime: 1520985084279741
actualTime: 1520985084279743
CRITICAL ERROR: Estimated Time can't be before observation time
estimateTime: 1520985097049459
actualTime: 1520985097049460

jgkamat, 2018-06-04:
Ill try to take a closer look if I get time next week but its not as important now as it shouldn't cause crashes and shouldn't cause very wrong computations.

When I looked last, it looked like the current time wasn't getting updated often enough for some reason in the ball filter.

On June 3, 2018 8:25:23 PM EDT, Matthew Woodward <notifications@github.com> wrote:
>@guyfleeman Soccer doesn't crash anymore but we were never able to
>figure out exactly what was causing the faulty values.

-- 
Sent from my Android device with K-9 Mail. Please excuse my brevity.

ashaw596, 2018-06-04:
That patch might probs fix it maybe. Can't actually run/reproduce it so not sure. Also not 100% why we were using the log time to begin with. Might break things. Using _lastTrackTime would probs also work

JNeiger, 2018-06-05:
Anybody remember the steps we had to do to reproduce it? I thought it was just shaking the ball around but that's not working on my VM.

ashaw596, 2018-06-05:
If that's where it's come from. I'd guess it's from when the simulator or
soccer lags out. I'd try running a complex play and shake the ball. Your
computer might be too powerful lol.

JNeiger, 2018-06-05:
I got it to fail once on master just letting the ball sit in the on the field for a few minutes. Haven't seen anything yet on your branch

jgkamat, 2018-06-18:
Unfortunately this is still an issue, but we only saw it once and I'm unable to reproduce it. We would need to get a new stacktrace to debug this further

JNeiger, 2019-01-18:
I haven't been able to replicate this since last June. I vote we close this and just reopen if someone else gets a new stack trace

closed 2019-01-18 by JNeiger