RoboJackets/robocup-software #1125: Compilation

maxaubel, 2018-04-28:
We are currently experiencing some troubles when trying to compile the most recent version.
This is what I try to "make"

[image]

maxaubel, 2018-04-28:
This happened after trying to update the geometries of the new field (square penalty area)

chachmu, 2018-04-29:
That can happen when you try to change things in the submodules. Try running:
 ```
make clean
git submodule update --init --recursive
```
If that doesn't work try rm'ing the directory and `git reset --hard`

maxaubel, 2018-04-29:
Yes it worked, thanks!

jgkamat, 2018-04-29:
Nice, ill close this, feel free to reopen if you run into more issues :)

closed 2018-04-29 by jgkamat