RoboJackets/robocup-software #1126: Can't bind to to the %%%% team's radio port

maxaubel, 2018-05-12:
We have this problem when trying to run an instance of soccer -sim -b, but then it says:

```"what():  Can't bind to the yellow team's radio port."```

[image]

jgkamat, 2018-05-13:
Yah, there actually is a bug where we connect to yellow anyway on startup, and then immediately switch over to blue. I really need to take a look at this, so this is probably a good enough excuse to :). 

To work around it, start the -b soccer version first, so it won't error out when (wrongly) connecting to the yellow team.

closed 2018-05-16 by jgkamat

maxaubel, 2018-05-17:
Is there a way to not require RX radio? I mean, without modifying the code a lot, send data without needing a response from robots? 
We had implemented it for a previous version, but it "died" (stop working, and won't assign behaviours of plays to robots because of "No RX") after some other changes (not staged). 
@jgkamat do you use telegram btw?

jgkamat, 2018-05-17:
I don't think there's a way to achive this easily, we use RX for quite a lot of things, such as knowing which robots are live, whether robots can kick, if they are in a healthy state, etc. The 'manual' mode should overrides this though, since it dosen't need all the data and is useful for debugging:

[image]

It only activates if it finds a joystick attached to the computer though.

That said, I'm pretty sure that we do send packets when robots are "visible". You might be able to hack that on by overriding [this if statement](https://github.com/RoboJackets/robocup-software/blob/master/soccer/Processor.cpp#L481), and a couple more in the same loop (`r->visible` is referenced many times in that loop to cut off non-visible robots).

 I think you might be running into another issue though:

>  and won't assign behaviours of plays to robots because of "No RX"

This can happen for a lot of reasons, (for example, if you have a behavior that requires a kicker, but the robot does not report it has a working kicker). Could I see the exact error message? 

If you could share the patch you had before, I could try to figure out why the patch is no longer working (that would be a lot easier than starting from scratch).

I'm not on telegram, but I am `jgkamat` on freenode. We also have a [gitter room](https://gitter.im/RoboJackets/robocup-software), although I check that much less often than IRC nowadays.

maxaubel, 2018-05-19:
Hey,
I will add you to our modification of your code then.We will be working on
bypassing the "No RX" RX now.

Thanks!

Sin otro particular, se despide cordialmente,

Maximiliano Aubel
Estudiante de Ingeniería Civil Informática
Universidad Técnica Federico Santa María

PR #1127 (2018-05-16): Fix initialization of simradio on yellow for a frame
Closes #1126, very simple.

@chachmu, can I get a quick review? :)