RoboJackets/robocup-software #1136: Kickoff makes our decision tree disappear

Jason27chan, 2018-06-06:


jgkamat, 2018-06-06:
Im a bit confused by this, do you mean the behavior tree? I tested a full game a couple of weeks ago and everything worked fine. Could you take some screenshots? Are you sure all the relevant plays are enabled? 

If there is an issue, the earliest ill be able to fix it is Friday.

On June 5, 2018 8:39:59 PM EDT, Jason Chan <notifications@github.com> wrote:
>Assigned #1136 to @jgkamat.

-- 
Sent from my Android device with K-9 Mail. Please excuse my brevity.

ashaw596, 2018-06-06:
Idk. But with real robots. The kickoff play completed before the kick.
Maybe it was a vision issue tho or something.

Jason27chan, 2018-06-10:
We can't reproduce this either. We'll worry about this when it comes up

closed 2018-06-10 by Jason27chan